cyberbotics/webots #2095: HingeJointParameters.position is increasing after position change

agayev169, 2020-08-07:
I am trying to do some reinforcement learning with Webots, and trying to make an episodic task, I wrote a reset function which would reset the position of a robot and other objects. The problem occurs when I reset the robot's position. I first tried to do this with E-puck robot and got the problem with wheels, when wheels change their position and robot started "flying". I thought that I should try the same with another robot, so I followed [tutorial 6](https://cyberbotics.com/doc/guide/tutorial-6-4-wheels-robot) to make a custom robot. After the creation of a robot, I faced the same problem again; however, this time I could watch the values of joints and other parts of a robot, and, as far as I understand, the position of a joint is not static and attached to a robot, but changes dynamically. Therefore, the solution to this problem is to reset the position of a joint every time I reset the position of a robot. This seems error-prone to me, and maybe it would be better to make the position of joints depending on the position of a parent object. Besides, as far as I understand, it is impossible to adjust this parameter for E-puck (and maybe other robots as well), since we do not have direct access to the joints (or maybe it is possible, I am new to Webots) which makes it a bug.

You can see what happens to a robot on the following [video](https://youtu.be/Vid-8Y-BIGM).

omichel, 2020-08-07:
How do you reset your robot?
You should not only reset the robot position (e.g, set the translation and rotation fields to their original values) but also the physics, e.g, call [wb_supervisor_node_reset_physics](https://www.cyberbotics.com/doc/reference/supervisor#wb_supervisor_node_reset_physics) on your robot node.
This should fix the problem.

agayev169, 2020-08-07:
> How do you reset your robot?
> You should not only reset the robot position (e.g, set the translation and rotation fields to their original values) but also the physics, e.g, call [...]

Yes, you are rights, I did not think that I may need to do this. Thank you very much!

closed 2020-08-07 by agayev169