cyberbotics/webots #2098: Python ImportError: dynamic module does not define module export function

JasonLrh, 2020-08-10:
On Mac 10.15.6, Webots 2020B, brew python@3.8.5
>Traceback (most recent call last):
  File "walk.py", line 2, in <module>
    from controller import Robot,Camera
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dynamic module does not define module export function (PyInit__controller)
WARNING: 'walk' controller exited with status: 1.  

and the env of mine is 
>$ echo $WEBOTS_HOME 
/Applications/Webots.app

When I use 2020a, I can go to the `Applications/Webots.app/lib/controller/python` to rebuild the lib. but in 2020b, this directory is no longer exist. How can I correctly import controller module?

ps: the front of my code is below
~~~python
#!/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/bin/python3.8
import os
import sys

for i in sys.path:
    print(i)
import cv2 as cv

libraryPath = os.path.join(os.environ.get("WEBOTS_HOME"), 'projects', 'robots', 'robotis', 'darwin-op', 'libraries',
                           'python37')
libraryPath = libraryPath.replace('/', os.sep)
sys.path.append(libraryPath)
from managers import RobotisOp2GaitManager, RobotisOp2MotionManager
from controller import Robot,Camera
~~~

DavidMansolino, 2020-08-10:
> When I use 2020a, I can go to the Applications/Webots.app/lib/controller/python to rebuild the lib. but in 2020b, this directory is no longer exist. How can I correctly import controller module?

You should simply add the python version at the end of the path.

JasonLrh, 2020-08-10:
> You should simply add the python version at the end of the path.  

but I can use python3(python3.8) directly in the Terminal. Does it means I've added the python in the PATH

DavidMansolino, 2020-08-10:
Yes, however on mac terminal PATH and Application PATH might differ, are you starting Webots from the terminal or from the icon?
In any case on macOS it is strongly recommended to set the full path to the python executable in the Webots preferences:
https://cyberbotics.com/doc/guide/preferences#general

JasonLrh, 2020-08-10:
> Yes, however on mac terminal PATH and Application PATH might differ, are you starting Webots from the terminal or from the icon?
> In any case on macOS it is strongly recommended to set the full path [...]

I add the acture python path in the head of PATH. Nothing changed. I've solved those problem in 2020a by rebuild them in `${WEBOTS_HOME}/lib/controller/python` but this directory disappear in 2020b.

The `_controller.so ` and `_vehicle.so ` do need to re-build for a slight platform changed. I've tried to re-build the `Webots` app but after generating there isn't a directory `${Webots_build_dir}/lib/controller/python*`

DavidMansolino, 2020-08-10:
I just checked with a fresh installation of R2020b and I have the ``/lib/controller/python*`` folders, if you don't have them, maybe you should try re-installing the application:
[image]

JasonLrh, 2020-08-10:
> I just checked with a fresh installation of R2020b and I have the `/lib/controller/python*` folders, if you don't have them, maybe you should try re-installing the application:
> [image]

yes. my folder is same. I also have `python[num][num]`, but what I need is a source directory `python`

DavidMansolino, 2020-08-10:
The sources are in `webots/resources/languages/python` (for the standard Webots Python API, but you will find similar path for extension (e.g. vehicle, darwin-op, etc.)).

JasonLrh, 2020-08-12:
OK,I build all Webots source, and it works well now. Thanks.

JasonLrh, 2020-08-12:
Done

closed 2020-08-12 by JasonLrh