cyberbotics/webots #2101: An error in the external controller of multi-process startup webots

NZS-YNU, 2020-08-10:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I started multiple python external controllers through multiple processes, but I found that not all external controllers can be started normally in webots, especially when I speed up the simulation of webots. Sometimes only 1 to 3 external controllers will be activated, and an error will be reported at this time: Only one instance of the Robot class should be created

**Screenshots**
[image]
[image]

**System**
 - Operating System: [Windows 10]
 - Graphics Card: [ NVIDIA GeForce 1660 6GB]

**Additional context**
（1）The multi-process program is as follows：
pool = multiprocessing.Pool(processes=4)
pool.apply_async(run_controller1, args=(True, 0.3, 0.3,))
pool.apply_async(run_controller2, args=(True, 0.3, 0.3,))
pool.apply_async(run_controller3, args=(True, 0.3, 0.3,))
pool.apply_async(run_controller4, args=(True, 0.3, 0.3,))
pool.close()
pool.join()
（2）The controller structure is as follows：
def run_controller1(is_reset, left_speed, right_speed):
    supervisor = Supervisor()
    robot_node = supervisor.getFromDef("robot1")
    trans_field = robot_node.getField("translation")

    if is_reset:
        y = trans_field.getSFVec3f()[1]
        inital = [params.robots_init_loc[0][0], y, params.robots_init_loc[0][1]]
        trans_field.setSFVec3f(inital)
        robot_node.resetPhysics()

    else:
    
        run_controller(supervisor, left_speed, right_speed)

        final_loc = trans_field.getSFVec3f()

        final_loc = [final_loc[0], final_loc[2]]

        with open(params.path_state_save[0], "w") as f:
            f.write(str(final_loc[0]) + " " + str(final_loc[1]))

omichel, 2020-08-10:
The error message you get is caused by instantiating twice the Robot (or Supervisor) class in a single Python controller. It doesn't come from Webots, but from the `CppController.dll` library which is linked against each Python controller and ensures only one instance of Robot/Supervisor is created per controller process. Are you sure you are not instantiating twice a Supervisor or Robot class in one of your controllers?

NZS-YNU, 2020-08-13:
> The error message you get is caused by instantiating twice the Robot (or Supervisor) class in a single Python controller. It doesn't come from Webots, but from the `CppController.dll` library which [...]

well, I reinstalled webots, the problem disappeared........

closed 2020-08-13 by NZS-YNU