cyberbotics/webots #2110: Test Suite Makes Webots Crashes on macOS

DavidMansolino, 2020-08-11:
**Describe the Bug**
The test suite is making Webots crashes on MacOS, here is the stack:
```
0   libsystem_kernel.dylib        	0x00007fff7144d33a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff71509e60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff713d4808 abort + 120
3   libGPUSupportMercury.dylib    	0x00007fff587000a5 gpusGenerateCrashLog.cold.1 + 95
4   libGPUSupportMercury.dylib    	0x00007fff586f718f gpusGenerateCrashLog + 89
5   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2f9baf28 gpusKillClientExt + 9
6   libGPUSupportMercury.dylib    	0x00007fff586f855b gpusSubmitDataBuffers + 164
7   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ec270a6 IntelCommandBuffer::getNew(GLDContextRec*) + 48
8   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ec26eb4 intelSubmitCommands + 171
9   GLEngine                      	0x00007fff41ec9395 gleUnbindTextureObject + 58
10  GLEngine                      	0x00007fff41e9b451 gleUnbindDeleteHashNameAndObject + 63
11  GLEngine                      	0x00007fff41dfb9a5 glDeleteRenderbuffersEXT_Exec + 599
12  com.cyberbotics.webots        	0x000000010d92dc6d wren::texturecubemapbaker::bakeBrdf(wren::ShaderProgram*, unsigned int) + 669 (TextureCubeMapBaker.cpp:360)
13  com.cyberbotics.webots        	0x000000010d92dd3e wr_texture_cubemap_bake_brdf + 30 (TextureCubeMapBaker.cpp:385)
14  com.cyberbotics.webots        	0x000000010d4f872f WbPbrAppearance::preFinalize() + 511 (WbPbrAppearance.cpp:113)
15  com.cyberbotics.webots        	0x000000010d5901cf WbShape::preFinalize() + 159
16  com.cyberbotics.webots        	0x000000010d41087e WbGroup::preFinalize() + 238 (WbGroup.cpp:60)
17  com.cyberbotics.webots        	0x000000010d63fb3f WbTransform::preFinalize() + 31 (WbTransform.cpp:56)
18  com.cyberbotics.webots        	0x000000010d5bfc2e WbSolid::preFinalize() + 94 (WbSolid.cpp:273)
19  com.cyberbotics.webots        	0x000000010d41087e WbGroup::preFinalize() + 238 (WbGroup.cpp:60)
20  com.cyberbotics.webots        	0x000000010d35668d WbBaseNode::finalize() + 445
21  com.cyberbotics.webots        	0x000000010d5a60ca WbSimulationWorld::WbSimulationWorld(WbProtoList*, WbTokenizer*) + 1354 (WbSimulationWorld.cpp:84)
22  com.cyberbotics.webots        	0x000000010d3a3895 WbControlledWorld::WbControlledWorld(WbProtoList*, WbTokenizer*) + 53 (WbControlledWorld.cpp:38)
23  com.cyberbotics.webots        	0x000000010d3a52e5 WbControlledWorld::WbControlledWorld(WbProtoList*, WbTokenizer*) + 37 (WbControlledWorld.cpp:81)
24  com.cyberbotics.webots        	0x000000010d327c1d WbApplication::loadWorld(QString, bool) + 5037 (WbApplication.cpp:335)
25  com.cyberbotics.webots        	0x000000010d486507 WbMainWindow::loadWorld(QString const&, bool) + 135 (WbMainWindow.cpp:1328)
26  com.cyberbotics.webots        	0x000000010d4704ef WbMainWindow::loadDifferentWorld(QString const&) + 31 (WbMainWindow.cpp:1307)
```

DavidMansolino, 2020-08-11:
Removing the ``--minimize`` argument in the test suite fixes the crash.

DavidMansolino, 2020-08-11:
Other crashes in the test suite on macOS:
```
0   libsystem_kernel.dylib        	0x00007fff7144d33a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff71509e60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff713d4808 abort + 120
3   libGPUSupportMercury.dylib    	0x00007fff587000a5 gpusGenerateCrashLog.cold.1 + 95
4   libGPUSupportMercury.dylib    	0x00007fff586f718f gpusGenerateCrashLog + 89
5   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2f9baf28 gpusKillClientExt + 9
6   libGPUSupportMercury.dylib    	0x00007fff586f855b gpusSubmitDataBuffers + 164
7   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ec270a6 IntelCommandBuffer::getNew(GLDContextRec*) + 48
8   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ec26eb4 intelSubmitCommands + 171
9   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ed51d9a gldPresentFramebufferData + 120
10  GLEngine                      	0x00007fff41dc48d1 glSwap_Exec + 97
11  com.apple.opengl              	0x00007fff41da6473 CGLFlushDrawable + 59
12  com.apple.AppKit              	0x00007fff348b575a -[NSOpenGLContext flushBuffer] + 20
13  libqcocoa.dylib               	0x00000001109bc8e8 0x110964000 + 362728
14  org.qt-project.QtGui          	0x00000001042e821c QOpenGLContext::swapBuffers(QSurface*) + 412
15  com.cyberbotics.webots        	0x0000000101dd7205 WbWrenWindow::renderNow(bool) + 357 (WbWrenWindow.cpp:231)
16  com.cyberbotics.webots        	0x0000000101dabeab WbView3D::renderNow(bool) + 219 (WbView3D.cpp:1413)
17  com.cyberbotics.webots        	0x0000000101dd7582 WbWrenWindow::event(QEvent*) + 130 (WbWrenWindow.cpp:251)
```

And:
```
0   org.qt-project.QtCore         	0x00000001066bb993 0x106616000 + 678291
1   org.qt-project.QtCore         	0x00000001066c7134 QString::compare_helper(QChar const*, int, char const*, int, Qt::CaseSensitivity) + 212
2   com.cyberbotics.webots        	0x00000001043ccfa7 QString::operator==(char const*) const + 71 (qstring.h:1236)
3   com.cyberbotics.webots        	0x0000000104739c2c WbViewpoint::followStringToType(QString const&) + 28 (WbViewpoint.cpp:678)
4   com.cyberbotics.webots        	0x00000001047311c4 WbViewpoint::followType() const + 36 (WbViewpoint.hpp:79)
5   com.cyberbotics.webots        	0x000000010474831c WbView3D::onSelectionChanged(WbAbstractTransform*) + 348 (WbView3D.cpp:258)
6   com.cyberbotics.webots        	0x000000010475827f QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<WbAbstractTransform*>, void, void (WbView3D::*)(WbAbstractTransform*)>::call(void (WbView3D::*)(WbAbstractTransform*), WbView3D*, void**) + 143 (qobjectdefs_impl.h:152)
7   com.cyberbotics.webots        	0x00000001047581e8 void QtPrivate::FunctionPointer<void (WbView3D::*)(WbAbstractTransform*)>::call<QtPrivate::List<WbAbstractTransform*>, void>(void (WbView3D::*)(WbAbstractTransform*), WbView3D*, void**) + 88 (qobjectdefs_impl.h:186)
8   com.cyberbotics.webots        	0x0000000104758125 QtPrivate::QSlotObject<void (WbView3D::*)(WbAbstractTransform*), QtPrivate::List<WbAbstractTransform*>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) + 165 (qobjectdefs_impl.h:415)
9   org.qt-project.QtCore         	0x0000000106831e4b QMetaObject::activate(QObject*, int, int, void**) + 1435
10  com.cyberbotics.webots        	0x0000000104869614 WbSelection::selectionChangedFromSceneTree(WbAbstractTransform*) + 84 (WbSelection.moc.cpp:209)
11  com.cyberbotics.webots        	0x000000010464bfcb WbSelection::clear() + 59 (WbSelection.cpp:147)
12  com.cyberbotics.webots        	0x000000010464cb34 QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbSelection::*)()>::call(void (WbSelection::*)(), WbSelection*, void**) + 132 (qobjectdefs_impl.h:152)
13  com.cyberbotics.webots        	0x000000010464caa8 void QtPrivate::FunctionPointer<void (WbSelection::*)()>::call<QtPrivate::List<>, void>(void (WbSelection::*)(), WbSelection*, void**) + 88 (qobjectdefs_impl.h:186)
14  com.cyberbotics.webots        	0x000000010464c9e5 QtPrivate::QSlotObject<void (WbSelection::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) + 165 (qobjectdefs_impl.h:415)
15  org.qt-project.QtCore         	0x0000000106831e4b QMetaObject::activate(QObject*, int, int, void**) + 1435
16  org.qt-project.QtCore         	0x0000000106829a73 QObject::~QObject() + 259
17  com.cyberbotics.webots        	0x000000010458da80 WbNode::~WbNode() + 1520 (WbNode.cpp:348)
18  com.cyberbotics.webots        	0x00000001044193ab WbBaseNode::~WbBaseNode() + 267 (WbBaseNode.cpp:66)
19  com.cyberbotics.webots        	0x0000000104736d3b WbViewpoint::~WbViewpoint() + 395 (WbViewpoint.cpp:170)
20  com.cyberbotics.webots        	0x0000000104736e65 WbViewpoint::~WbViewpoint() + 21 (WbViewpoint.cpp:170)
21  com.cyberbotics.webots        	0x0000000104736e8c WbViewpoint::~WbViewpoint() + 28 (WbViewpoint.cpp:160)
22  com.cyberbotics.webots        	0x000000010456aefd WbMFNode::clear() + 205 (WbMFNode.cpp:66)
23  com.cyberbotics.webots        	0x00000001044d4713 WbGroup::deleteAllChildren() + 35 (WbGroup.cpp:134)
24  com.cyberbotics.webots        	0x000000010466a678 WbSimulationWorld::~WbSimulationWorld() + 328 (WbSimulationWorld.cpp:137)
25  com.cyberbotics.webots        	0x0000000104468537 WbControlledWorld::~WbControlledWorld() + 791 (WbControlledWorld.cpp:103)
[... cut ...]
27  com.cyberbotics.webots        	0x000000010ce3e63c WbControlledWorld::~WbControlledWorld() + 28 (WbControlledWorld.cpp:83)
28  com.cyberbotics.webots        	0x000000010cdbe293 WbApplication::~WbApplication() + 67 (WbApplication.cpp:106)
29  com.cyberbotics.webots        	0x000000010cdbe5f5 WbApplication::~WbApplication() + 21 (WbApplication.cpp:116)
30  com.cyberbotics.webots        	0x000000010cdbe61c WbApplication::~WbApplication() + 28 (WbApplication.cpp:104)
31  com.cyberbotics.webots        	0x000000010cf1c2dc WbMainWindow::closeEvent(QCloseEvent*) + 300 (WbMainWindow.cpp:1091)
```

DavidMansolino, 2020-08-11:
This crash is linked to the `supervisor_reset_simulation` test (reproducible outside of the test suite, in fast and run mode only):
```
0   libsystem_kernel.dylib        	0x00007fff7144d33a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff71509e60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff713d4808 abort + 120
3   libGPUSupportMercury.dylib    	0x00007fff587000a5 gpusGenerateCrashLog.cold.1 + 95
4   libGPUSupportMercury.dylib    	0x00007fff586f718f gpusGenerateCrashLog + 89
5   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2f9baf28 gpusKillClientExt + 9
6   libGPUSupportMercury.dylib    	0x00007fff586f855b gpusSubmitDataBuffers + 164
7   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ec270a6 IntelCommandBuffer::getNew(GLDContextRec*) + 48
8   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ec26eb4 intelSubmitCommands + 171
9   com.apple.driver.AppleIntelKBLGraphicsGLDriver	0x00007fff2ed51d9a gldPresentFramebufferData + 120
10  GLEngine                      	0x00007fff41dc48d1 glSwap_Exec + 97
11  com.apple.opengl              	0x00007fff41da6473 CGLFlushDrawable + 59
12  com.apple.AppKit              	0x00007fff348b575a -[NSOpenGLContext flushBuffer] + 20
13  libqcocoa.dylib               	0x00000001109bc8e8 0x110964000 + 362728
14  org.qt-project.QtGui          	0x00000001042e821c QOpenGLContext::swapBuffers(QSurface*) + 412
15  com.cyberbotics.webots        	0x0000000101dd7205 WbWrenWindow::renderNow(bool) + 357 (WbWrenWindow.cpp:231)
16  com.cyberbotics.webots        	0x0000000101dabeab WbView3D::renderNow(bool) + 219 (WbView3D.cpp:1413)
17  com.cyberbotics.webots        	0x0000000101dd7582 WbWrenWindow::event(QEvent*) + 130 (WbWrenWindow.cpp:251)
```

brettle, 2024-08-19:
When this was last investigated, were the macos test_suite crashes reliably reproducible or were they intermittent?

omichel, 2024-08-19:
As far as I remember, this was systematic.

closed 2024-08-31 by brettle