cyberbotics/webots #2114: rostopic echo /......./...../recognition_objects doesnt work: no message

SherlockSheep, 2020-08-12:
Hi. First I met the problem the same as [#1637](https://github.com/cyberbotics/webots/issues/1637) with webots 2020a, then I upgraded webots to 2020b.
This time when I try to echo /..../...../recognition_objects, the datatype/md5sum error's gone, but there's no message at all. Other topics works fine.
This is my screenshot, what i got by rostopic hz /..../...../recognition_objects.
[image]

SherlockSheep, 2020-08-12:
[image]
Publisher exists.
And bounding boxes in webots' camera overlay looks fine.
Guys please help me.

DavidMansolino, 2020-08-12:
I can indeed confirm the issue, we will have a look at it.

DavidMansolino, 2020-08-12:
As a temporary workaround I just found that the `../camera/recognition_objects` topic is published if you subscribe to the `../camera/image` topis too.

SherlockSheep, 2020-08-12:
@DavidMansolino You are right, it works.
Thank you very much!!  : )

PR #2115 (2020-08-12): Fix ROS camera recognition topic
**Description**
Fix #2114
closed 2020-08-12 by DavidMansolino