cyberbotics/webots #2116: Wrong time stamp for ros topic

SherlockSheep, 2020-08-12:
I'm using webots 2020b now which I installed by .deb package. And I'm using ROS kinetic + Ubuntu 16.04.
When I echo the topics published by webots node, I noticed that the time stamps are abnormal.

For example, I launched one of the tutorial projects by:
`roslaunch webots_ros pioneer3at.launch`

When the simulation finished starting up, I echo header of gps topic, and I got:
[image]

Then I echo another topic's header, I got:
[image]

Another one:
[image]

Another one:
[image]

It seems that the time stamps ware frozen at some random value.
It's critical for me because in my project I need time stamp for tf lookup.

DavidMansolino, 2020-08-12:
I just checked it and for me, it seems to work fine, both topics are publishing the same header stamp:

[image]

If you add the `--clock` and `--use-sim-time` argument to the ros controller in Webots (https://www.cyberbotics.com/doc/guide/using-ros#using-the-standard-ros-controller) and set the `use_sim_time` ros param (http://wiki.ros.org/rosparam#rosparam_set) to true, then the simulation time is used:
[image]

DavidMansolino, 2020-08-12:
Same when using the launch file (`roslaunch webots_ros pioneer3at.launch`):

[image]

jhonasiv, 2020-08-13:
+1. I am having the same issue using ROS Noetic and Ubuntu 20.04. I tried using sim time but it didn't solve it.

SherlockSheep, 2020-08-13:
> +1. I am having the same issue using ROS Noetic and Ubuntu 20.04. I tried using sim time but it didn't solve it.
> 
> EDIT: just tried with Webots2020A and it worked.

I tried uninstalling and reinstalling 2020b, and it still doesn't work. : (
I am now trying 2020a.

DavidMansolino, 2020-08-13:
@SherlockSheep ok, let us know if it works on R2020a for you too.
P.S. @jhonasiv and @SherlockSheep which version of ROS are you using?

SherlockSheep, 2020-08-13:
@DavidMansolino  I just rollback to 2020a_rev1 and it works fine. I'm using ROS kinetic and Ubuntu 16.04 LTS.

DavidMansolino, 2020-08-13:
Ok thank you for the test. And are you using the webots_ros package compiled from source or installing the pre-compiled package?

SherlockSheep, 2020-08-13:
I'm use webots_ros package which I installed by 'sudo apt-get install ros-kinetic-webots-ros'.

DavidMansolino, 2020-08-13:
Ok thank you for this information. Last question, is it up to date?

SherlockSheep, 2020-08-13:
I suppose the answer is yes, I got 2.1.0-1.

DavidMansolino, 2020-08-13:
That's indeed the latest version.

SherlockSheep, 2020-08-13:
Althouge In 2.1.0-1 the world files are 2020b style and some protos are not supported by 2020a. The topic's time stamp works fine.

DavidMansolino, 2020-08-13:
Ok then the problem is certainly at the ros controller level in Webots.
We will try to fix this soon.

jhonasiv, 2020-08-13:
> @SherlockSheep ok, let us know if it works on R2020a for you too.
> P.S. @jhonasiv and @SherlockSheep which version of ROS are you using?

I am using ROS Noetic and tried using webots_ros both source compiled and the pre-compiled package to no avail.

DavidMansolino, 2020-08-14:
I can finally reproduce it with this procedure:
   - Get Webots R2020b from the tarball and extract it.
   - Set `WEBOTS_HOME` to point to the extracted tarball folder.
   - Install the debian `ros-kinetic-webots-ros` package.
   - Start the launch file: `roslaunch webots_ros pioneer3at.launch`.
   - Echo two topic headers:
      - `rostopic echo /pioneer3at/gps/values/header`
      - `rostopic echo /pioneer3at/accelerometer/value/header`

DavidMansolino, 2020-08-14:
This is unfortunately not reproducible in the development environment.

DavidMansolino, 2020-08-14:
Even when creating the package locally and installing it, it works fine.
It only fails when the package is created on travis and then installed on another machine.
One explanation could be the fact that the ros header changed and we are using old headers in our dependencies.

SherlockSheep, 2020-08-16:
I tried webots 2020b (still installed from .deb) today on a newly installed Ubuntu 18.04 with ROS melodic. The problem still exists. So I have two choice:
1. Try compiling webots from code, and see if the issue still exist.
2. Stay with 2020a temporarily until 2020b-rev1 comes out.

DavidMansolino, 2020-08-17:
Or another alternative:
    3. Wait that we finalize the fix (should be ready in maximum a few days: https://github.com/cyberbotics/webots/pull/2127) and then use a nightly build of R2020b-rev1.

closed 2020-08-24 by DavidMansolino