cyberbotics/webots #2124: R2020b: Webots Dusk and Night themes distort toolbar placement on Ubuntu 18.04.

sowani, 2020-08-13:
**Describe the Bug**
On Ubuntu 18.04 in R2020b, when Theme is changed to "Webots Night" or "Webots Dusk", placement of toolbars gets disturbed.

**Steps to Reproduce**
1. Start Webots and ensure that Theme is set to "Webots Classic". The toolbars should appear as in screenshot "webots_classic_theme.png" file.
2. Go to Tools -> Preferences
3. Click on Themes dropdown and select "Webots Night" theme.
4. Press "OK".
5. Webots will get restarted.
6. Check the placement of toolbars.
7. Repeat step 2-6 with theme "Webots Dusk".

**Expected behavior**
Toolbars should be left justified, as seen in the "Classic" theme. However, the toolbars get shifted to right, leaving lot of empty space before them. This can be clearly seen in "webots_night_theme.png" and "webots_dusk_theme.png" files.

**Screenshots**
[image]
[image]
[image]

**System**
 - Operating System: Linux Ubuntu 18.04.5 LTS
 - Graphics Card: None.

**Additional context**
None.

stefaniapedrazzi, 2020-08-13:
This is made on purpose and it is part of the theme style.
In the classic theme the toolbar is left justified. 
In dusk and night themes the toolbar is centered.

sowani, 2020-08-13:
Okay... In that case this issue can be closed.

closed 2020-08-13 by sowani