cyberbotics/webots #2128: Running model on webots_2020b fails while it runs great on webots_2020a

HefnySco, 2020-08-16:
**Describe the Bug**
A clear and concise description of what the bug is.
I use this [world](https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/examples/Webots/worlds/webots_quadX.wbt)  
**Steps to Reproduce**
1. run Webots_2020b
2. open World https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/examples/Webots/worlds/webots_quadX.wbt
3. clean and recompile controllers.
4. The drone is not stable and it falls while taking off.
=================
The same procedures runs successfully on Webots_2020a

Another thing is that if I run Webots_2020a but with controllers compiled under Webots_2020b it also falls from the sky. 

**System**
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NA

stefaniapedrazzi, 2020-08-20:
Could you provide a simple simulation where the issue is reproducible but that doesn't depend on your specific libraries?

A simple simulation will help you and us to identify the issue and it will make possible to fix it earlier.

HefnySco, 2020-08-20:
I suggest it has to do with timing. Quad tends to do like that when correction to motors is too slow.
now there is the otherthing... if you open webots2020a again -after you run webots 2020b- and run the same simulation without recompiling any plugins using webots 2020a and used the compiled plugins that were compiled by webots2020b it will acts the same as webots2020b. i.e. webots2020a will act the same as webots2020b.
Please check this third video https://youtu.be/9Ahch27aRJY

HefnySco, 2020-08-25:
guys anyfeedback. do you want extra demos. how can I help ?

DavidMansolino, 2020-08-25:
I am sorry but without a simplified version of your simulation (without external dependencies) it is extremely hard and time consuming for us to understand the issue.

HefnySco, 2020-08-26:
Hey... Finally solved.
The issue is that I used to depend on "northDirection" and look for values [1,0,0] for local maps and [0,0,1] for maps that are created using "World generated using the Open Street Map to Webots importer".
And accordingly adjust coordinates of GPS &  Interia sensors.

For local maps the value retured from reading "northDirection" was not correct in the new version as the property has been removed. That is the main cause. 
Anyway It has been solved for local maps and works very well, for open street map I fixed it with a hack for timebeing.

Many Thanks  for your great product. I luv it.

DavidMansolino, 2020-08-26:
That perfectly makes sense, thank you for the feedback!

closed 2020-08-26 by DavidMansolino