cyberbotics/webots #2131: Supervisor services for a robot throwing errors on running rosservice call

koyalbhartia, 2020-08-18:
**Describe the Bug**
Supervisor services for a robot throws communication related errors on running rosservice call through the terminal. This is tested on a robot with Supervisor set to TRUE in the Webots Simulator. 

**Steps to Reproduce**
1. Make a repository containing the package webots_ros and build it.
2. In the "worlds" folder within webots_ros, create a world with a robot and the robot's Supervisor field set to TRUE, and set the Controller of the robot as "ros".
3. Start ros either using "roscore" or running a launch file. 
4. Open the webots world created in step 2, and make sure that the robot has connected to ROS in the console.
5. From a new terminal, source on the working repo. i.e source devel/setup.bash.
6. Type "rosservice list". This will output the services related to Supervisor and also other services.
7. Pick any of the Supervisor service, and try to do a rosservice call on it using rosservice call /service_name args., it throws an error.
8. The consistent behavior is as follows: Eg: If the value of my arg(field:) is 0, I get **ERROR: service [/<robot_name>/supervisor/field/set_string] responded with an error: b''.**
9. For any other value of the arg, I get **ERROR: transport error completing service call: unable to receive data from sender, check sender's logs for details.**
10. Eg: Run:
rosservice call /robot_5170_koyal/supervisor/field/set_vec3f "field: 0
index: 0
value:
  x: 1.0
  y: 0.0
  z: 0.0"

or Run:
rosservice call /robot_5170_koyal/supervisor/field/set_vec3f "field: 1
index: 0
value:
  x: 1.0
  y: 0.0
  z: 0.0"

**or any other Supervisor service with the required arguments.**

**Expected behavior**
Expected behavior is that the rosservice call on any of the Supervisor service should perform it's functionality instead of throwing the above mentioned and below shown errors.

**Screenshots**
[image]

This behavior is consistent with all Supervisor related services.

**System**
 - Operating System: Linux Ubuntu 20.04.1 LTS, Lenovo Thinkpad P73
- Processor : Intel® Core™ i7-9750H CPU @ 2.60GHz × 12 
 - Graphics: Quadro P620/PCIe/SSE2 / Quadro P620/PCIe/SSE2

stefaniapedrazzi, 2020-08-19:
Do you get the same errors if you use valid values?
For example:
1. Get the node
    ```
      rosservice call /robot_5170_koyal/supervisor/get_self "ask: false"
    ```
     -> you should receive the node id (something like ``value: 140121294782976``)

2. Get the field
    ```
    rosservice call /robot_5170_koyal/supervisor/node/get_field "node: 140121294782976
    fieldName: 'translation'
    proto: false" 
    ```
    please replace the node value by the valued you get in step 1
     -> you should receive the node id (something like ``field: 140121294780800``)

3. Set the field value
    ```
    rosservice call /robot_5170_koyal/supervisor/field/set_vec3f "field: 140121294780800
    index: 0
    value:
    x: 1.0
    y: 2.0
    z: 3.0" 
    ```
    please replace the field value by the valued you get in step 2
     -> you should receive ``success: 1``

I tested on Ubuntu 18.04 with melodic and it seems to work for me with valid values.
If I use the same invalid values as in the description, then I get the the same errors.
We should try to improve the error description, but it seems correct that it throws errors.

koyalbhartia, 2020-08-20:
Thank you for your reply Stephani!

[image]

As you can see in the above screenshot, I ran the **get_self service** and received a value of **16530896** as expected.

Next as you can see I did the rosservice call on **supervisor/node/get_field**; here are the observations.
1. When I press tab to autocomplete and complete the expected args so that I could type in the required args, the **proto option does not appear.** 
2. Hence I continued and replaced the **node value with 16530896 and fieldName as translation.**
3. This gave the following **error:** Looks like a communication issue related to md5sum.
**ERROR: unable to connect to service: remote error reported: client wants service /robot_6704_koyal/supervisor/node/get_field to have md5sum ebc69c91abb4723f5b25baca5ecfe051, but it has 80105ddcb86dd98488ad3cf099686b86. Dropping connection.**

4. Next I again ran the same service and **manually filled in the proto field** too as suggested in your previous comment and as you can see in the screenshot below. 

[image]

And as you can see there's a lot of error.

**What I think is the problem!**
This last error looks like a problem with Python. I am using **Ubuntu 20.04 and ROS Noetic** in which it's a mandate to use Python3. However, I think your ROS controller is written in Python2.7, and hence the above issue.

stefaniapedrazzi, 2020-08-20:
Ok thank you, these error messages are more relevant.

I will also check with ROS Noetic.

stefaniapedrazzi, 2020-08-20:
Which version of Webots are you running?
If you don't have the `proto` argument option, this is probably because you are running Webots < R2020b.
I would suggest you to install the latest version R2020b and the latest webots-ros package for noetic so that you can be sure that webots-ros and Webots are compatible.

DavidMansolino, 2020-08-20:
>This last error looks like a problem with Python. I am using Ubuntu 20.04 and ROS Noetic in which it's a mandate to use Python3. However, I think your ROS controller is written in Python2.7, and hence the above issue.

Unfortunately this is not the reason, because the ros controller in Webots is written only in C++.

But I think the problem is that you are using the latest version of the `webots_ros` package with an old version of Webots. In Webots R2020b we introduced a new argument to the `node/get_field` service, this has probably changed the MD5 of the service, therefore if you have updated the webots_ros package you should update the version of Webots you are using too.

koyalbhartia, 2020-08-20:
[image]

As shown in the above screenshot, my webots is at version R2020b.
And as seen in the below screenshot, all these packages are successfully installed and updated.

[image]

DavidMansolino, 2020-08-21:
As discussed on the Discord server, it appeared that the local 'srv' file in the workspace were outdated.

closed 2020-08-21 by DavidMansolino