cyberbotics/webots #2139: Noise becomes fixed patterned.

SherlockSheep, 2020-08-24:
Set noise field of Camera node to any positive value, for example 0.2, and start simulation. At the beginning image noise works fine, which is random noise in 3 channels: 
[image]

But minutes later(simulation time), the image noise becomes fixed patterned, and is only black-and-white:
[image]

Lidar noise suffers from the same issue.

This issue exists on both 2020a-rev1 and 2020b-rev1-nightly-build. I tested with ubuntu 16.04 and ROS kinetic.

DavidMansolino, 2020-08-24:
I just tried and for me, it seems to work fine (after 1hour of simulation time):

[image]

What is your GPU? Can you give us the output of the 'webots --sysinfo' command?

SherlockSheep, 2020-08-24:
$ webots --sysinfo
System: Linux 4.15.0-107-generic x86_64
Processor: Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz
Number of cores: 8
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: GeForce RTX 2080/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 440.100

SherlockSheep, 2020-08-24:
By the way, I'm trying to make a transparent wall to test my sonar application, which should be seen by Lidar. I made a `Solid` with Box `Shape`, and I set `appearance PBRAppearance` > `transparency` to 1. But the wall was still detected by Lidar. How could I accomplish my purpose?

DavidMansolino, 2020-08-24:
Your configuration looks perfect. Can you identify an exact time or action that makes the noise looks like this?

To make the wall completely transparent, you should completely remove the Shape node.

SherlockSheep, 2020-08-24:
Thank you! I shall post a screen recording video later.

SherlockSheep, 2020-08-24:
@DavidMansolino 
Hi, I've post a video on youtube:
https://youtu.be/Yhu3B104F0M

DavidMansolino, 2020-08-26:
Thank you for the video, I tried many times and unfortunately was not able to reproduce the issue.
Can you please try with a nightly build of Webots R2020b-rev1: https://github.com/cyberbotics/webots/releases

SherlockSheep, 2020-08-26:
I tested Webots Nightly Build (25-8-2020) with the computer mentioned above (Ubuntu16.04+ROS Kinetic), the issue existed.
Then I tested with another computer (Ubuntu18.04+ROS Melodic), and the issue also existed.
This is the configuration of the second computer:
```
System: Linux 5.4.0-42-generic x86_64
Processor: AMD Ryzen 5 3600 6-Core Processor
Number of cores: 12
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: GeForce RTX 2060/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 440.100
```
Could it be relevant to OpenGL version?

DavidMansolino, 2020-08-27:
Just to make sure I tried with the exact same OpenGL version:
```
System: Linux 5.4.0-42-generic x86_64
Processor: Intel(R) Core(TM) i7-7500U CPU @ 2.70GHz
Number of cores: 4
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: GeForce 940MX/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 440.100
```

But I  still can't reproduce the problem, I don't have a lot more ideas to understand where does the problem come from. Do both of your computers use the same GPU?

SherlockSheep, 2020-08-27:
No they are not the same. One of them is RTX 2080, the other one is RTX 2060.

SherlockSheep, 2020-08-28:
@DavidMansolino 
Hi David, it seems that this issue is related to the GPU we are using.
My colleague helped trying on another computer, which has GTX 1080 and the OpenGL version is also 4.6.0. The noise worked perfectly fine.
So RTX 20 series are (maybe) to blame.

DavidMansolino, 2020-08-28:
Hi @SherlockSheep, ok that's good that you were able to test with another kind of GPU, it seems to make sense now.
Unfortunately, I don't have any RTX 20.. GPU available to test right now, but if I am not wrong my colleague @stefaniapedrazzi has one and will be able to test this next week.

SherlockSheep, 2020-08-28:
Thank you for your patient response. I have another question to ask: must Webots run with graphical interface? We want to use Webots on our server for CI testing. If Webots could run without graphical interface, there would be no bother installing one.

DavidMansolino, 2020-08-28:
> I have another question to ask: must Webots run with graphical interface? We want to use Webots on our server for CI testing. If Webots could run without graphical interface, there would be no bother installing one.

It's a bit tricky, but yes, Webots can run without graphical interface, we use CI too and run Webots inside, we use a virtual framebuffer (XVFB) to run it.
We run it on Github Actions for example: https://github.com/cyberbotics/webots/blob/master/.github/workflows/test_suite_linux.yml#L63-L90
We also have an example using Travis if needed.

DavidMansolino, 2020-08-28:
Note that we also plan to create a Github Action that will automatically install Webots in your Github Action environment to ease the use of Webots in GitHub Action, would this be useful for you?

SherlockSheep, 2020-08-29:
Thank you! That's exactly what we need! : )

stefaniapedrazzi, 2020-09-02:
I can reproduce the issues on my linux system with an RTX 2060 GPU.

Even at the beginning the noise is sometimes a fixed black and white pattern instead of random RGB (every 10/30 simulation steps). But when running the simulation the fixed pattern appears always more frequently.

stefaniapedrazzi, 2020-09-07:
I suspect the issue comes from some floating point over-/under-flow handled differently in RTX GPUs and used to compute the pseudo random values.

What changed in RTX NVidia Turing GPUs is that they also support half floating point precision to speed up computations.
I'm now testing to find a way to fix it.

stefaniapedrazzi, 2020-09-07:
Here is a post presenting a modified noise implementation that doesn't use the `sin` function and works correctly on my system.
http://byteblacksmith.com/improvements-to-the-canonical-one-liner-glsl-rand-for-opengl-es-2-0/

DavidMansolino, 2020-09-07:
That looks good, I just tested on my computer (GTX 970) and works fine too.

SherlockSheep, 2020-09-07:
Thank you @stefaniapedrazzi : )

PR #2215 (2020-09-09): Fix noise generation in GLSL
Fix #2139: fix generation of pseudo random values in GLSL shaders that didn't properly work on RTX Nvidia GPUs.
closed 2020-09-09 by stefaniapedrazzi