cyberbotics/webots #2147: Can i integrate my android application with webots simulator created?

i-am-vyshu, 2020-08-26:
Hi Team

I have just started with webots and i am planning to simulate a robot cleaner (domestic robot) using webots. I have an android application that is used to pair with the physical robot and control the functionality. 

My question, can i use this webot simulator to connect/integrate with my android application so that i can perform the functionalities on the simulated robot from the android application.

DavidMansolino, 2020-08-26:
If you plan to run Webots on a desktop computer and use your android application to communicate with the controller of the simulated robot, yes, this is for sure feasible.

omichel, 2020-08-26:
For this kind of questions, please use our Discord channel, not GitHub.

closed 2020-08-26 by omichel