cyberbotics/webots #2157: Segfault on load when camera view is configured in a pop-out window

WasabiFan, 2020-08-31:
**Describe the Bug**

My custom PROTO with a named camera causes Webots to crash upon loading the world if the camera view is configured to be popped out in a separate window.

**Steps to Reproduce**

`Demo.proto`:
```
#VRML_SIM R2020b utf8

PROTO Demo [
  field  SFVec3f     translation     0 0 0
  field  SFRotation  rotation        0 1 0 0
  field  SFString    controller      "void" # Is `Robot.controller`.
  field  MFString    controllerArgs  []     # Is `Robot.controllerArgs`.
  field  SFString    customData      ""     # Is `Robot.customData`.
  field  SFBool      supervisor      FALSE  # Is `Robot.supervisor`.
  field  SFBool      synchronization TRUE   # Is `Robot.synchronization`.
  field  SFBool      selfCollision   FALSE  # Is `Robot.selfCollision`.
]
{
  Robot {
    translation IS translation
    rotation IS rotation
    controller IS controller
    controllerArgs IS controllerArgs
    customData IS customData
    supervisor IS supervisor
    synchronization IS synchronization
    selfCollision IS selfCollision
    children [
      Camera {
        name "mycamera"
      }
    ]
  }
}
```

`demo.wbt`:

```
#VRML_SIM R2020b utf8
WorldInfo {
  info [
    "Demo world"
  ]
  title "Demo"
  basicTimeStep 16
  coordinateSystem "NUE"
}
Viewpoint {
  orientation 0.04858061663146304 0.9695925120162909 0.23985471504568262 3.0288380896471887
  position 0.3664507544999487 1.0747987087308764 -2.0504698100019554
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Floor {
  size 100 100
}
Demo {

}
WoodenBox {
  translation 0.605448 0.2 -0.00121984
  size 0.6 0.4 0.6
}
```

Open the above project. Double-click the black camera view so it appears in a new window. Hit the "reload" button to reload the world, or alternately close and re-open Webots.

The application appears, begins to render, and shows the pop-out for a short time, before crashing with a segfault.

**Expected behavior**

Webots would load without crashing.

**Screenshots**
[image]

**System**
Pop! OS 20.04.

```
$ webots --version
Webots version: R2020b
$ webots --sysinfo
System: Linux 5.4.0-7634-generic x86_64
Processor: Intel(R) Core(TM) i7-4930K CPU @ 3.40GHz
Number of cores: 12
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: GeForce GTX 780 Ti/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 440.100
```

**Additional context**
I don't know what I am doing, and am blindly hacking on PROTO files until it does what I want. This is the most minimal repro I could reasonably produce. I didn't see an obvious way to generate a core dump, but if that'd be helpful and someone can let me know what the proper procedure would be I'd be happy to get that done.

omichel, 2020-08-31:
I believe this bug is already fixed [here](#2119). Can you try if you can reproduce it with a recent [nightly build](https://github.com/cyberbotics/webots/releases) of Webots?

WasabiFan, 2020-08-31:
You guys are fast! :grin: That did indeed fix the problem.

For others' reference, I downloaded the R2020b-rev1 nightly:

```
Webots version: R2020b revision 1 Nightly Build 28/8/2020 83fdfeede46351e591fb6fe200ba94ca45838ef7
```

Thanks!

closed 2020-08-31 by WasabiFan

WasabiFan, 2020-08-31:
Actually, that fixes the crash, but introduces (a variety of?) other bug(s) regarding the camera. Inexactly:
- The camera image in the popout window is black unless I first start the simulation _then_ open the separate window
- Splitting it into a separate window causes rendering issues in the main preview

[image]

The former, at least, appears to be mentioned in #2119 as a supposed bugfix. Is this something I should be investigating more thoroughly and opening issues for or is this expected?

stefaniapedrazzi, 2020-08-31:
I will try to reproduce the new issues you mentioned.
Could you please write step by step the actions you do?

WasabiFan, 2020-08-31:
The world which exhibited the above symptoms is not the same one as the simplified version I included in the OP; I won't be able to sit down and make it a reproducible sample until tomorrow (~16hrs from now). That being said, for our purposes I suspect the original world can be reduced to just a camera with a controller that enables it on load.

Broadly, the steps are:
- Pop out the camera view
- Re-load the project
- Observe the camera view is now black.

stefaniapedrazzi, 2020-08-31:
I tested both in Ubuntu 18.04 and Ubuntu 20.04 with the built-in camera example (`samples/devices/worlds/camera`) and unfortunately I cannot reproduce the two issues:
even after reloading the world, the camera image is shown in the external window and the main view rendering is correct.

We would need some more details:
- which OS are you using?
- how did you install Webots?
- is the robot window always open? or when do you open it?
- do you run the simulation before reloading the project?
- is the simulation running when you reload the simulation?
- do you reload or reset the simulation?

stefaniapedrazzi, 2020-08-31:
It would also be very helpful if you could try to reproduce the same issues on another linux installation (preferably Ubuntu).