cyberbotics/webots #2160: Documentation link handling doesn't respect "ctrl-click" to open in new tab

WasabiFan, 2020-08-31:
**Documentation page(s) concerned**
Most user guide and reference pages. For example: https://www.cyberbotics.com/doc/reference/inertialunit

**What is the problem with the current documentation?**
Clicking on links to other pages, for example using the left nav or on an inline link for another node, while holding <kbd>Ctrl</kbd> does not open in a new tab. Normally, the browser would handle this, but the docs pages override the click handler. The manual handling of the link then doesn't check the original event to see if it should be opening in a new tab. Relevant event handler:

https://github.com/cyberbotics/webots/blob/0ed61bd0390dc98de84f64b3a1692069893a61cc/docs/js/viewer.js#L264

**Describe how you would fix it**

I don't know the original motivation for the JS link handling, so this might not solve the original problem, but I assume the event's default action would work fine:

```js
if (event.ctrlKey) {
    return;
}
```

omichel, 2020-08-31:
The motivation for this mouse click handling is to avoid reloading the full page and only get the content corresponding to the new link (and update the URL in the navigation bar). We should indeed handle the <kbd>Ctrl</kbd> case explicitly...

closed 2020-08-31 by omichel

WasabiFan, 2020-08-31:
Thanks for the updates!

Would it also make sense to do this for intra-page anchor links? (https://github.com/cyberbotics/webots/blob/master/docs/js/viewer.js#L240) With the above change it makes links to another page open in a new tab but links within the page don't. The `IntertialUnit` node page linked above has a handful of these.

PR #2174 (2020-08-31): Fix Ctrl-click in documentation pages
Fixes #2160.
Test it here: https://www.cyberbotics.com/doc/reference/inertialunit?version=fix-ctrl-click-in-doc