cyberbotics/webots #2164: [ROS]Time step service cannot be looped

Bingsheng1991, 2020-08-31:
**Describe the Bug**
I'm trying to control multiple epucks in Webos using ROS with changing wheel speeds. This requires time_step service to be called for each iteration in order to update actuator data. However, the service can only be called once (only the first speed takes affect), then it stops after one iteration.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Update wheel speeds continuously under ROS

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Ubuntu 18.02
 - Graphics Card: Nvidia 2080 Super
 - Webots Version: R2020b

**Additional context**
Add any other context about the problem here.

DavidMansolino, 2020-09-03:
I just tried with a world with two e-pucks, everything seems to work as expected:
1. start roscore and the simulation => simulation runs
2. call once the step service for first e-puck: `rosservice call /epuck1/robot/time_step "{ value: 32 }"` => now simulation stops because robot 1 is waiting for new call to this service.
3. call the service every 1 second: `while true; do rosservice call /epuck1/robot/time_step "{ value: 32 }"; sleep 1; done` => now simulation runs again.
4. call once the step service for second e-puck: `rosservice call /epuck2/robot/time_step "{ value: 32 }"` => now simulation stops because robot 2 is waiting for new call to this service.
5. call the service every 1 second: `while true; do rosservice call /epuck2/robot/time_step "{ value: 32 }"; sleep 1; done` => now simulation runs again the service is called for both robots.

omichel, 2020-11-30:
I am closing this issue. Feel free to re-open if needed.

closed 2020-11-30 by omichel