cyberbotics/webots #2179: Pop-out camera preview black after world reload

WasabiFan, 2020-09-01:
**Describe the Bug**

The pop-out camera preview works initially, but becomes irreversibly blank upon project reload.

[image]

I have also occasionally seen the broken buffer wrapping partially shown previously but it doesn't have a consistent repro.

**Steps to Reproduce**

This is via the following demo project, and DOES NOT seem to appear on the stock camera examples:

[demo.zip](https://github.com/cyberbotics/webots/files/5153599/demo.zip)

Repro steps:

- Open the provided world
- Observe that the camera preview is displayed
- Double-click the camera preview
- Note that the camera preview is still displayed
- Reload the project (without closing Webots)
- The camera view is now black.

**Expected behavior**

To see the live camera preview like is shown within the original window.

**Screenshots**

Included above.

**System**
Pop! OS 20.04.

```
$ webots --version
Webots version: R2020b
$ webots --sysinfo
System: Linux 5.4.0-7634-generic x86_64
Processor: Intel(R) Core(TM) i7-4930K CPU @ 3.40GHz
Number of cores: 12
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: GeForce GTX 780 Ti/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 440.100
```

**Additional context**

Follow-up to separate issue after discussion in #2157. 

I don't have a stock Ubuntu environment handy (neither the will to install natively nor an OpenGL-compatible VM). Minus the window manager tweaks, my pop! environment should be equivalent to an Ubuntu 20.04 system with the proprietary NVIDIA drivers enabled. That being said, naturally, it's possible it's something nuanced about my environment.

omichel, 2020-09-01:
I cannot reproduce this bug on Windows with the latest nightly build.

omichel, 2020-09-01:
I cannot reproduce it on Linux either (using the current master version).
@WasabiFan: may I ask you to try with the latest [nightly build](https://github.com/cyberbotics/webots/releases) (August 31st)?

stefaniapedrazzi, 2020-09-02:
I can now reproduce the issue with your demo simulation.
The key point is to use the camera recognition functionality.

The same problem is also reproducible with the `projects/samples/devices/worlds/camera_recognition.wbt` simulation.

PR #2196 (2020-09-07): Notify when the overlay foreground texture is updated
Fix #2179: changes to the overlay foreground texture are not notified to the `WbRenderingDeviceWindow`.

`WbRenderingDevice::foregroundTextureIdUpdated()` is defined but never used.
But if the external camera window has to be shown at load (or reload), the foreground texture is reinitialized after the camera window was setup and thus the window still links to an invalid texture id.
The foreground texture is only used in case camera recognition.
closed 2020-09-07 by stefaniapedrazzi