cyberbotics/webots #2229: ENU Coordinate System Next Steps

DavidMansolino, 2020-09-11:
As discussed in #1643, to finish the ENU feature, here are the missing steps:
  - [ ] Implement #1638 and in particular the support for `EXTERNPROTO`.
  - [ ] Convert more worlds to ENU (currently only one is converted): #1835.
  - [ ] Convert more FLU PROTO nodes.
  - [ ] Fix web export (in particular the display of Background cube maps in X3D views for ENU worlds), e.g. implement `wren.js`.
  - [ ] Update controllers (e.g the ros controller is not doing the ENU conversion internally)
  - [ ] Set the new worlds and projects by default to ENU.
  - [ ] Improve the OSM exporter to export worlds in the ENU format.
  - [ ] Update the test suite worlds to ENU.
  - [ ] Fix the remaining FIXME in the background node: https://github.com/cyberbotics/webots/blob/53b9967abac7d7b3ad2f684c75afdb4a2a747aff/src/webots/nodes/WbBackground.cpp#L345
  - [ ] Update simulations in other repos:
     - [x] https://github.com/cyberbotics/webots_ros2:  cyberbotics/webots_ros2#123
       - [ ] Cleanup the forked enu PROTO nodes such as: https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_epuck/protos
     - [ ] https://github.com/cyberbotics/webots_ros
  - [ ] Update the URDF importer.
  - [ ] Update the Blender exporter.

mrbhjv, 2020-12-15:
Hi,
is this still projected to be on R2021a.rev1? As a ROS user, it'd be very useful to have the matching coordinate system.

omichel, 2020-12-16:
Yes, we will try. Any help on that front is welcome 😉.

lukicdarkoo, 2020-12-16:
> Hi,
> is this still projected to be on R2021a.rev1? As a ROS user, it'd be very useful to have the matching coordinate system.

Since R2020b, you can use the ENU coordinate system. This issue is about the next steps that will make the ENU coordinate system easier to use and more transparent for users. For example, Webots worlds in `webots_ros` and `webots_ros2` repositories are in the ENU coordinate system.

Is there anything on the list that you are particularly interested in?

mrbhjv, 2020-12-17:
Sorry, I didn't explain correctly, AFAIK, if you try to import any predefined object (at least most of them), they come with non-ENU coordinate systems. So when we began to use the simulator, we didn't quite understand why sometimes either our model or the world were rotated, even though we explicitly changed the coordinate systems to ENU. So, our floor would be _vertical_ and anything we added would just fall (or our model would always be rotated). So, as you said, it's more about being able to use them in a more transparent way.
A colleague mentioned to me there's a branch we could build Webots from, that has most of the objects/worlds already using ENU as the base coordinate system. I'm just bumping for it to be on the mainstream :). So that we can use all the base nodes or base protos, without having to make some bridge for the coordinate systems (which is what I understand would need to be done now).
Maybe my understanding is wrong but I hope this explains better what I meant.

lukicdarkoo, 2020-12-17:
I see. The objective is indeed to convert all PROTOs to ENU. A very few PROTOs is converted to ENU because we don't see it as urgent. 

For non-robot PROTOs (e.g. arena, fridge, house...) it should not be an issue, you always apply the same rotation? You usually have to change the default rotation in any case. 

> A small tip here, use <kbd>ALT</kbd>+<kbd>1</kbd> to get a top view of your world.

The problem may be related to the internal coordinate system of the robot models. Depending on the application, it may be necessary to use the ENU coordinate system in the robot models. In that case, you need to change the robot model by yourself.

Here is a script that may help you:
https://github.com/cyberbotics/webots/tree/master/scripts/converter

BenjaminDeleze, 2021-11-15:
Replaced by #3073

closed 2021-11-15 by BenjaminDeleze