cyberbotics/webots #2234: Mismatch between local and recorded web animation

yuchen93, 2020-09-11:
**Describe the Bug**
Recorded html5 animation is different from the simulation. Object position lags behind moving gripper.

**Steps to Reproduce**
I created a html5 recording of a simulation but the animation is different from the local simulation. 

**Expected behavior**
In local simulation, everything looks normal. However, in the animation replay, when the gripper moves objects, the object's position lags behind the gripper's position. 

**Screenshots**
[image]
[image]
[image]

**System**
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce RTX 2080 11 GB

**Additional context**
I do have a lot of objects in the environment, that might be the issue since I had no problem with a different environment with less object. I'd like to know if there is an easy fix (possibly from the html end) for my environment. 

Thanks

stefaniapedrazzi, 2020-09-14:
Do you get this issue after jumping to a specific time during the animation replay or also when just starting the animation without moving manually the time slider?

Could you please share your simulation and your generated animation files (html, json, and x3d files)?

yuchen93, 2020-09-21:
I get this without jumping time, you can see the html replay here: https://www.cs.utexas.edu/~yuchen93/trash-sorting-sim/html/trash_sorting_arena.html

Attached the project files here as well. You can find the created html, json and x3d in there. I removed the texture files to reduce file size. 

Thanks a lot for looking into this!

[belt_sorting_sim.zip](https://github.com/cyberbotics/webots/files/5257156/belt_sorting_sim.zip)

yuchen93, 2020-09-22:
I tried to dynamically load the objects to reduce the number of total moving object but the resulting animation now does not even have the dynamically loaded objects.
[image]

stefaniapedrazzi, 2020-09-23:
> I tried to dynamically load the objects to reduce the number of total moving object but the resulting animation now does not even have the dynamically loaded objects.

Adding objects dynamically is not supported in HTML animation mode.
Another option is to initially locate the object somewhere else on the floor but not on the conveyor belt (this is slowing down the simulation and increasing the size of the animation JSON file) and move objects on the conveyor belt only when needed.

I can reproduce the issue with your world and I'm now inspecting it.

stefaniapedrazzi, 2020-09-23:
It is not the position of the objects (cans bottles) that is wrong, but the links of the UR robot that are no longer correctly aligned when the robotics arm moves.
This occurs with this particular controller but seems to work fine with the default `ure_can_grasper` controller.

yuchen93, 2020-09-23:
OK! I'll look into fixing my controller. This issue does not show up in the local simulation though, why is that the case?

stefaniapedrazzi, 2020-09-24:
There is clearly a bug somewhere when exporting the solid position. I'm still working on it.

stefaniapedrazzi, 2020-09-28:
This bug should be fixed in #2266.
The patch will be included in the nightly build of Webots R2020b rev 2 starting from tomorrow (https://github.com/cyberbotics/webots/tags)

PR #2266 (2020-09-28): Compute Animation step update after Solid coordinates have been updated
**Description**
Fixes #2234: the animation recording system publishes the changes too early, before the Solid nodes translation and rotation field values are correctly updated.

This issue alone won't create so big position/rotation mismatches if it was not for the `WbSimulationCluster::step` that updates the Solid coordinates earlier (in the middle of the physics step) in order to update the DistanceSensor rays.
This should also be inspected, because the translation values computed when updating the rays are different fro [... cut]
closed 2020-09-28 by stefaniapedrazzi