cyberbotics/webots #2236: UR10e collision bug

Simon-Steinmann, 2020-09-12:
There is a collision bug, when using self collision. The objects do not touch where it shows the collision points. I created a world showing the example. 
[ur10e_collisionbugwbt.zip](https://github.com/cyberbotics/webots/files/5212731/ur10e_collisionbugwbt.zip)

[image]

DavidMansolino, 2020-09-14:
There is indeed something wrong going out with this simulation, I can reproduce the issue too:

[image]

DavidMansolino, 2020-09-14:
I tried simplifying the bounding objects to keep only the two 'colliding' cylinders (to make sure it was not a collision with a 'hidden' bounding object), but the problem is still here:
[image]

westinsykes, 2020-09-25:
As a note, real UR arms treat that pose similarly to a collision in order to prevent people's fingers from getting pinched.

omichel, 2020-10-02:
Fixed in #2282.

closed 2020-10-02 by omichel