cyberbotics/webots #2241: Can't get contact info when using a track.

zapplelove, 2020-09-16:
**Describe the Bug**
I have a simulation which is a robot running on a track, so I need to get contact values when a robot foot touches down on the track. But when it happens, the foot of the robot has turned red but no value I get. When I use the floor to replace the track, I get the touchdown value 1.

**Steps to Reproduce**
1. here is a description of the robot foot:
[image]
[image]

2. this is the results when using a track:
[image]

3. this is the results when using a Floor:
[image]

**Expected behavior**
when the robot touches a track belt, I want to get the return value 1 using the touchsensor.

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce RTX 1050

DavidMansolino, 2020-09-16:
I tried to reproduce the bug in the provided 'bumber.wbt' simulation by adding a ConveyorBelt node in this world, but unfortunately in this simulation it works fine for me.
Can you maybe just for trying replace the bounding object of the touch sensor with a box?
And also try displaying the contact points in both cases (from the 'View=>Optional Rendering' menu)?

zapplelove, 2020-09-16:
> I tried to reproduce the bug in the provided 'bumber.wbt' simulation by adding a ConveyorBelt node in this world, but unfortunately in this simulation it works fine for me.
> Can you maybe just for [...]

thanks for your reply. I will try the provided 'bumber.wbt' simulation and find the differences between it and my robot proto file. If I solve this, I will reply to you.

DavidMansolino, 2020-09-16:
Perfect thank you.

zapplelove, 2020-09-16:
> Perfect thank you.

Thanks again! I try the `bumber .wbt` simulation and it does work! So I change the shape of my foot touch sensor to a simple sphere, and set the radius a little big than the real, it can work well too. Maybe the shape size really matters.

closed 2020-09-16 by zapplelove