cyberbotics/webots #2242: Feature request: support buttons or other simple graphical UI inputs

WasabiFan, 2020-09-16:
**Is your feature request related to a problem? Please describe.**

My controller has a mode I'd like to be able to enable/disable. The keyboard interface (or even mouse or joystick) could work, but the buttons aren't discoverable unless I document them myself.

**Describe the solution you'd like**

A class similar to `Keyboard` or `Mouse` which allows creation of labelled buttons which appear somewhere in the Webots UI; probably in their own tab in the robot window. An API on this class would then allow to check if the button has been pressed since last tick. A "toggle" button which shows its state would also be nice.

**Describe alternatives you've considered**

As mentioned above, `Mouse`, `Keyboard` and `Joystick` are fairly natural choices here in lieu of an on-screen button.

**Additional context**

N/A

stefaniapedrazzi, 2020-09-16:
Did you consider using a simple robot window to enable/disable your controller mode graphically?
Because this is the typical functionality provided by the robot window.

You can find an example of a simple robot window in sample simulation `samples/howto/worlds/custom_html_robot_window.wbt`.

omichel, 2020-09-16:
The documentation for robot windows is available here: https://cyberbotics.com/doc/guide/controller-plugin#robot-window
Another good example is provided for the Thymio2 robot (see documentation about it).

WasabiFan, 2020-09-16:
I was _not_ aware of this feature and I'll have to experiment with it this evening before I can say if it meets my needs. It certainly _sounds_ like what I want!

WasabiFan, 2020-09-16:
Yeah, this seems to fit the bill.

closed 2020-09-16 by WasabiFan