cyberbotics/webots #2245: New mode: realtime without rendering

PymZoR, 2020-09-16:
**Is your feature request related to a problem? Please describe.**
Running webots headless in a CI environment with xvfb-run triggers some performances issues. As rendering is not required, It would be great to completely disable it (as in `fast` mode).

**Describe the solution you'd like**
A mode without rending (like `fast`) in 1x speed (like `realtime`)

**Describe alternatives you've considered**
Digging the reasons behind the performances drop while using xvfb-run

**Additional context**
Run context :
- Host on CentOS / intel CPU / nvidia GPU
- Gitlab CI runner inside a virtual machine
- Job executed in an ubuntu 20.04 based docker container (inspired by https://github.com/cyberbotics/webots-docker/blob/master/Dockerfile)

omichel, 2020-09-16:
I believe we should take the opportunity of this feature request to split the `mode` option in two:
Currently `--mode` option can take the following values: `pause`, `realtime`, `run` and `fast`.
My proposal would be to introduce a `--disable3dview` option and remove the `fast` option.
Therefore, to get the previous `--mode=fast` behavior, we would have to combine `--mode=run --disable3dview`.
And the new requested behavior would be obtained with `--mode=realtime --disable3dview`.
Similarly on the GUI, we should remove the fast button and replace it with a checkbox entitled "3D View" that would be checked by default.
What do you think about it?

DavidMansolino, 2020-09-16:
That looks like a very good idea to me!
Simple and flexible.

lukicdarkoo, 2020-11-09:
Implemented in #2286

closed 2020-11-09 by lukicdarkoo