cyberbotics/webots #2247: Inertia import and visualization requirements, and improve the simulation speed.

zapplelove, 2020-09-17:
**Is your feature request related to a problem? Please describe.**
### Feature request
+ I think the inertia of the robot has a big impact on the dynamics, but I have a complex simulate situation that I need the self-collision to get the force in which the touch sensor is in the robot, so I can't use the shape of the solid as the boudingObject(This will collide with the sensor and produce incorrect results. ). I need to use numerical values to give the inertia of the robot's member, but I don't know if that's right value. So I need a feature like Gazebo to see the inertia of the link.
+ Integrating `urdf2webots` functions into software and regenerate urdf file or sdf file.
+ The icon for Centroid visualization is too small.
+ When I simulate using the self-collision option, it is too slow.

omichel, 2020-09-18:
> So I need a feature like Gazebo to see the inertia of the link.

Can you post a screenshot of what you mean here?

> The icon for Centroid visualization is too small.

Do you mean the cross displayed when activating `View / Optional Rendering / Show Center of Mass` is too thin?

zapplelove, 2020-09-18:
> Can you post a screenshot of what you mean here?

In the Gazebo, you can use the `View/Center of Mass` and `inertias` to see if you set the right value. Here are some screenshots of my robot in the Gazebo:

> 
[image]
+ the original view of the robot foot link.
[image]
+ the view of the foot link with its inertia.
[image]
+ the view of the foot link with its center of mass.
> Do you mean the cross displayed when activating `View / Optional Rendering / Show Center of Mass` is too thin?

yes, it is.

omichel, 2020-09-18:
Simply clicking on your robot (selecting it) should display the bounding objects used to compute the inertia:
[image]

zapplelove, 2020-09-18:
> Simply clicking on your robot (selecting it) should display the bounding objects used to compute the inertia:
> [image]

Yes, I know this. But if I give the inertia instead of `boundingObject`, it would display nothing in webots

omichel, 2020-09-18:
> Yes, I know this. But if I give the inertia instead of boundingObject, it would display nothing in webots

I am not sure if it is possible to compute a inertia volume from an inertia matrix... Are you sure Gazebo uses only the inertia matrix to display these magenta boxes?

zapplelove, 2020-09-18:
> I am not sure if it is possible to compute a inertia volume from an inertia matrix... Are you sure Gazebo uses only the inertia matrix to display these magenta boxes?

I think it is. Because if I modify the value in the `<inertial><inertia>` label, the box will change, but in webots, I need to change `boundingObject`, actually, sometimes, I don't want the collision in the body but just the inertia.

omichel, 2020-09-18:
> I think it is. Because if I modify the value in the <inertial><inertia> label, the box will change

That looks weird... Are you sure this inertia tag doesn't contain any bounding object, just the inertia matrix? And are you sure the magenta box is only computed from the inertia matrix?

zapplelove, 2020-09-20:
> > I think it is. Because if I modify the value in the  label, the box will change
> 
> That looks weird... Are you sure this inertia tag doesn't contain any bounding object, just the inertia matrix? And are you sure the magenta box is only computed from the inertia matrix?

The urdf2webots project has solved this problem, and It can use `inertia` label in the urdf file to import `inertiaMatrix` directly into webots, although it's not visualizable.

zapplelove, 2020-09-20:
I still have a problem. If I have a force sensor in a link like this:
[image]
How to add a `touchsenor` to get the right value of it? Or any solution to this?

zapplelove, 2020-09-20:
> I still have a problem. If I have a force sensor in a link like this:
> [image]
> How to add a `touchsenor` to get the right value of it? Or any solution to this?

There is a `positionsensor` of the joint, why not a `forcesensor` of it?

zapplelove, 2020-09-20:
> [image]

I try this method:
+ set the `boundingObject` of a `Group` of `Cylinder`s with different `Translation`s
+ add a `SliderJoint` and set `minStop/maxStop`'s value of 0.001;
+ add a `endPoint Solid`;
+ add a `TouchSensor` and set the `boundingObject` of a `Cylinder` with a `Translation`;
after these steps, a collision can occur between the touchsensor and the link. Then set the selfCollision to be true of the robot, but this breaks the joint, So this is not the right solution to this situation.

stefaniapedrazzi, 2020-09-21:
Maybe what you are looking for is the `Motor` force/torque feedback measurements that you can get from the Motor API:
https://www.cyberbotics.com/doc/reference/motor#wb_motor_enable_force_feedback

I you also want to add the force sensor shape on the model, then you could add it simply as part of the `endPoint Solid` node.

omichel, 2020-09-22:
> Thanks. I will have a try. And another question, I use one same controller to control the models created by webots and Adams separately, but the simulation results of them are quite different. Which result should I trust?

That's difficult to say without an in depth study of your Adams and Webots models. Are you sure that both models are the same? If so, the main difference probably comes from the fact that Webots and Adams don't use the same physics engine.

zapplelove, 2020-09-22:
> That's difficult to say without an in depth study of your Adams and Webots models. Are you sure that both models are the same? If so, the main difference probably comes from the fact that Webots and Adams don't use the same physics engine.

I'm sure about that, they use the same models and the same control program. I agree about the different physics engines cause different results.

zapplelove, 2020-09-22:
> Maybe what you are looking for is the `Motor` force/torque feedback measurements that you can get from the Motor API:
> https://www.cyberbotics.com/doc/reference/motor#wb_motor_enable_force_feedback [...]

I have used this method. Here are the results:
+ When I set the motor position, the force feedback the right value.
+ When I set the motor force, the force feedback is very close to zero.
Actually, in the real world, a force sensor in the body link will feedback a correct force value both position control and force control.

omichel, 2020-09-22:
As explained in the documentation you are referring to, the force feedback does not measure the force (resp. torque) specified with the wb_motor_set_force (resp. wb_motor_set_torque) function. Therefore you have to add it manually to get the total force applying there in order to compare to the motor position control case.

closed 2021-11-15 by BenjaminDeleze