cyberbotics/webots #2248: Robot Window Initialization Failure

cagrikaymak, 2020-09-17:
Hey everyone, I cannot open Robotis Op2's robot window on Windows 10. I get the following error: Error: robot window initialization failed
Error: Cannot load the "C: /Users/Mechatronics/AppData/Local/Programs/Webots/projects/robots/robotis/darwin-op/plugins/robot_windows/robotis-op2_window/robotis-op2_window.dll" robot window library. 
 If controller is written in C++, there is no problem. Everything is ok in Ubuntu, I only have this problem in Windows 10.
So I had to work on Ubuntu for a while, is there a solution for Windows 10? 
Thanks in advance
By the way, I am using Webots 2020b revision1 version. Python version is 3.7.3

DavidMansolino, 2020-09-18:
Thank you for this bug report, I can indeed reproduce this on Windows with the ``humanoid_marathon`` simulation.

cagrikaymak, 2020-09-18:
I'm looking forward. Many thanks in advance

omichel, 2020-09-18:
May be related with #1311.

stefaniapedrazzi, 2020-10-15:
Sorry for the late answer..
To fix your issue you should probably just copy the following `runtime.ini` file in your Python controller folder:
https://github.com/cyberbotics/webots/blob/master/projects/robots/robotis/darwin-op/controllers/motion_player/runtime.ini

This is fixing the issue for the `humanoid_marathon` simulation.

Please let us know if it is working for you as well.

cagrikaymak, 2020-10-16:
Many thanks, everything is ok now.

Stefania Pedrazzi <notifications@github.com>, 15 Eki 2020 Per, 15:33
tarihinde şunu yazdı:

closed 2020-10-16 by DavidMansolino