cyberbotics/webots #2253: Simulation running without step() calls

Simon-Steinmann, 2020-09-21:
**Describe the Bug**
When having an <extern> controller, and webots is waiting for it to connect, the simulation is running, even though no robot.step() calls are being made. This is especially apparent when being in "run" mode, which uses all available resources. In my opinion, the simulation should only get calculated, when there is actual simulation steps

**Steps to Reproduce**
1. Start a world containing a robot with <extern> controller
2. Set webots to 'run' mode
3. See/hear computer max out one cor
4. Set webots to 'realtime' or pause, see/hear core usage drop

stefaniapedrazzi, 2020-09-21:
This issue occurs only with extern controllers and not with controllers started by Webots.

Simon-Steinmann, 2020-09-22:
That is correct, but it is very bothersome, when working with extern controllers, especially for Reinforcement Learning, where running it at max speed is the default. So anytime you have webots open, and adjust your code, your system is running at maximum boost, wasting power, being noisy etc.

stefaniapedrazzi, 2020-09-28:
Yes, sure.
I just added some details that will complete the issue description and will help understanding it.

PR #2377 (2020-10-27): Fix active loop waiting for extern controllers
Fix #2253: Webots uses too many computer resources when waiting for extern controllers (especially in run/fast mode).

This fix consists in pausing the timer that triggers the ``WbControlledWorld::step()`` if Webots is waiting for an extern controller and restart the timer when an extern controller connects.
This way the step function is not continuously called and aborted.

Another change of this PR is that a user clicks on the step button while waiting for an extern controller first connection, the step request is ignored and the step button is  [... cut]
closed 2020-10-27 by stefaniapedrazzi