cyberbotics/webots #2254: Ubuntu 16.04: issues with nodejs during compilation

stefaniapedrazzi, 2020-09-21:
When compiling Webots `resources/web/wwi` on Ubuntu 16.04 docker image following the instruction on the wiki page (https://github.com/cyberbotics/webots/wiki/Docker#create-your-own-image) the compilation fails because of nodejs issues.

On Ubuntu 18.04 docket image it works correctly.

We should fix it or explicitly mention that only Ubuntu 18.04 and Ubuntu 16.04 are supported.

DavidMansolino, 2020-09-21:
What is the exact issue when compiling on Ubuntu 16.04? For me it compiles fine on native Ubuntu 16.04 (and should therefore be possible to compile on Docker).
In any case, I think these instructions (https://github.com/cyberbotics/webots/wiki/Docker#create-your-own-image) should be updated to use the new scripts (maybe we have to add a few dependencies in it): https://github.com/cyberbotics/webots/blob/master/scripts/install/linux_optional_compilation_dependencies.sh
This will avoid duplication and make the creation of the development environment simpler.

stefaniapedrazzi, 2020-09-21:
These are the errors:
[image]

nodejs is not properly set with the given instructions.

stefaniapedrazzi, 2020-09-28:
I will adjust the installation scripts so that they can be reused also in the docker image.

PR #2271 (2020-10-05): Enhancement Docker installation instructions
**Description**
Fix #2254: instructions to compile Webots on Docker container doesn't work on Ubuntu 16.04

**Tasks**
  - [x] Fix instructions in wiki page: https://github.com/cyberbotics/webots/wiki/Docker
  - [x] Adjust installation scripts so that they work on a Docker container
  - [x] Fix instructions to run the streaming server in a Docker container:
    * https://www.cyberbotics.com/doc/guide/web-simulation?version=enhancement-docker-installation#using-docker
    * https://github.com/cyberbotics/webots/wiki/Docker#run-streaming-server-in-dock [... cut]
closed 2020-10-05 by stefaniapedrazzi