cyberbotics/webots #2257: Broken Link https://cyberbotics.com/buy

nicholascgilpin, 2020-09-21:
**Documentation page(s) concerned**
E.g., for example https://cyberbotics.com/doc/reference/distancesensor#sonar-sensors
https://cyberbotics.com/buy 

**What is the problem with the current documentation?**
A clear and concise description of what's wrong with the current documentation.
The link to https://cyberbotics.com/buy is broken and leads to a 404 page from the Github repository

**Describe how you would fix it**
A clear and concise description of what you want to happen.

Ensure a page exists for that link on https://cyberbotics.com/
Edit the repository website description field on Github to point to the correct page.

stefaniapedrazzi, 2020-09-21:
The wrong link is set here:
https://github.com/cyberbotics/webots/blob/2c8d0ed5b6192374c7069e8f29513e65c281b9a7/.github/FUNDING.yml#L9

omichel, 2020-09-22:
It should now be fixed.

closed 2020-09-22 by omichel