cyberbotics/webots #2258: Docker Webots And Web Stream Options Not working

nezihsunman, 2020-09-21:
I am developing a web system that works on docker with django+ angular. In this system, I could not run the webot web stream settings I tried locally on the system docker. This system works with local webots settings but not working in docker image.

**First error:**  
On docker image
https://hub.docker.com/layers/cyberbotics/webots/R2020b-rev1-ubuntu18.04/images/sha256-3aa965bb490f03fbdcd566066393608e5935a19917b6348048ddb97de09b1d7a?context=explore

Run command: 
docker run --rm -it --env="DISPLAY" --env="QT_X11_NO_MITSHM=1" --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" -p 1234:1234 -p 8888:8888 cyberbotics/webots:R2020b-rev1-ubuntu18.04

xvfb-run --auto-servernum webots/webots --mode=fast --stdout --stderr --minimize --batch --stream="port=1234;disableTextStreams" webots/projects/robots/softbank/nao/worlds/nao_demo.wbt

**See error:**

ERROR: Cannot initialize the sound engine: Cannot initialize OpenAL default device 'OpenAL Soft'

**Second Error** 

For following docker file:

RUN apt install nano psmisc
RUN pip install websocket-client tornado nvidia-ml-py psutil requests
ENV AUDIODEV=null
ENV DEBIAN_FRONTEND=noninteractive
ENV DISPLAY=:99
ENV LIBGL_ALWAYS_SOFTWARE=true
ENV WEBOTS_HOME=/usr/local/webots
ENV QTWEBENGINE_DISABLE_SANDBOX=1
ADD server/config/simulation/video.json /webots/resources/web/server/config/simulation/video.json
ADD server/key/localhost /webots/resources/web/server/key/localhost
ADD server.sh /server.sh
RUN chmod 654 /server.sh
CMD ["/server.sh", ""]

For server.sh
#!/bin/sh
Xvfb :99 -screen 0 1024x768x16 &
cd $WEBOTS_HOME/resources/web/server
./server.sh start
while true; do sleep 1000; done

**Run command:** 
docker run -d -p 1999:1999 -p 2000:2000 -p 2001:2001 -it <docker image id>

**System**
 - Operating System: [Windows 10, Docker]
 - Graphics Card: [Nvideo geforce 960M]

[image]
[image]

stefaniapedrazzi, 2020-09-23:
Did you try to let docker publish the port 1234, that you are using for the Webots stream, to the host?
i.e. running
```
docker run -d -p 1234:1234 -it 
```

Port 1999, 2000, and 2001 are the default ports used by the streaming session and server scripts to automatically start Webots on request, but it doesn't seem that you are these scripts.

nezihsunman, 2020-09-23:
please correct me if i am wrong, i'm new to webot before this i just physically used naaqi.

`docker run --rm -it --env="DISPLAY" --env="QT_X11_NO_MITSHM=1" --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" -p 1234:1234 -p 8888:8888 cyberbotics/webots:R2020b-rev1-ubuntu18.04`

This code blog works, but I think the problem is the lack of documentation or configuration of different systems.

I solved the stream problem last night. With "--auto-servernum --server-num=1"  settings.

`xvfb-run --auto-servernum --server-num=1 webots --mode=fast --stdout --stderr --minimize --batch --stream="port=1234;disableTextStreams" projects/robots/softbank/nao/worlds/nao_demo.wbt`

Additionally, following docker script still not working. In the future, I want to automate this entire system on the web server and manage more than one webot container system.

RUN apt install nano psmisc
RUN pip install websocket-client tornado nvidia-ml-py psutil requests
ENV AUDIODEV=null
ENV DEBIAN_FRONTEND=noninteractive
ENV DISPLAY=:99
ENV LIBGL_ALWAYS_SOFTWARE=true
ENV WEBOTS_HOME=/usr/local/webots
ENV QTWEBENGINE_DISABLE_SANDBOX=1
ADD server/config/simulation/video.json /webots/resources/web/server/config/simulation/video.json
ADD server/key/localhost /webots/resources/web/server/key/localhost
ADD server.sh /server.sh
RUN chmod 654 /server.sh
CMD ["/server.sh", ""]

For server.sh
#!/bin/sh
Xvfb :99 -screen 0 1024x768x16 &
cd $WEBOTS_HOME/resources/web/server
./server.sh start
while true; do sleep 1000; done

But other error continues
**See error:**
ERROR: Cannot initialize the sound engine: Cannot initialize OpenAL default device 'OpenAL Soft'

stefaniapedrazzi, 2020-09-23:
> This code blog works, but I think the problem is the lack of documentation or configuration of different systems.

Here you can find the available documentation for the simulation streaming:
https://www.cyberbotics.com/doc/guide/web-simulation
What documentation are you missing?
The support of Docker image is quite new and the streaming server functionality is still a work in progress.
The configuration of the session and server scripts is very specific to the application so it is not straightforward to provide a solution working out-of-the-box.

> But other error continues
> See error:
> ERROR: Cannot initialize the sound engine: Cannot initialize OpenAL default device 'OpenAL Soft'

This is not very important for the streaming. This library is just used for playing the speaker, motor and collision sound but in case of streaming or docker images this error can simply be ignored.

stefaniapedrazzi, 2020-09-28:
Are the errors reported here related to #2267?

nezihsunman, 2020-09-28:
No, this is a new error for typescript web framework. Webots min js works
for pure javaquerry frameworks like REACTJS and not working angularJS.
That's why I am converting my projects from angular js to reactjs.

stefaniapedrazzi, 2020-09-28:
Ok, thank you.
I'm now fixing the instructions to run a streaming server in a Docker container-

stefaniapedrazzi, 2020-10-01:
I updated the instructions to run the simulation server on localhost in a Docker container, see:
https://www.cyberbotics.com/doc/guide/web-simulation?version=enhancement-docker-installation#using-docker

Please let me know if this is working for you as well or if you think that some information are still missing from the documentation.

closed 2020-10-05 by stefaniapedrazzi