cyberbotics/webots #2267: Webot.min.js with lack of Jquery build

nezihsunman, 2020-09-24:
**Describe the Bug**

I'm developing a web framework. During this development, I can get the files that the webot streams with index.html. Using standard javascript jquerry.
But the framework I developed uses Angular frontend django backend.

I want to manually download the script line below and embed it in ts. The most basic way normally used is to run the jquerry structure globally. To be able to import and use Webots .view structure globally.
But during this process, the error in the webot.min.js file is this ".
$(...).menu is not a function"

This is normally view for working backend javascript with 
`<script src="https://www.cyberbotics.com/wwi/R2020a/webots.min.js"></script>`
`view = new webots.View(playerDiv, mobileDevice);`
[image]

But when the error came to making this system running on angular, webot.min.js was not built correctly with https://github.com/cyberbotics/webots/tree/master/resources/web/wwi source code.

**See error:**
[image]

With following code block:
`declare const webots: any` in component

External Js file added to project with path angular/nego-web-frontend/src/assets/js/webot.min.js

Angular Js script is following:
 "scripts": [
              "./node_modules/jquery/dist/jquery.min.js",
              "./node_modules/bootstrap/dist/js/bootstrap.min.js",
              "src/assets/js/webot.min.js"
            ],

**Error `** is inside the webot.min.js

Connect code in side .component.ts: 

`connect() {
    let playerDiv = document.getElementById('playerDiv');
    this.view = new this.webot.View(playerDiv, this.mobileDevice);
    this.view.broadcast = true;
    this.view.open('ws://localhost:1234');
    // this.view.open('ws://' + this.ipInput.value + ':' + this.portInput.value);
    this.connectButton.value = 'Disconnect';
    this.connectButton.onclick = this.disconnect;
    this.ipInput.disabled = true;
    this.portInput.disabled = true;
  }`

**_To summarize the problem briefly, while the min.js file was being built, only backend-javascript was required to work and min.js was shrunk only with the "babel-minify": "^ 0.5.1" library in the build process. In order for this system to work in current structures, it must be mounted in a different minify structure running on current frontend systems using libraries such as the webpack library._**

omichel, 2020-09-28:
I not sure to fully understand your problem, but I believe that #2270 should address it.
Meanwhile, can you propose a solution to this problem? This would allow me to better understand what you actually need.

nezihsunman, 2020-09-28:
Thanks for all the supports.

The main problem is that the web image is controlled and run on the stream server on the website.

The webot.min.js file you provide for this is normally working in a javascript base project. In this project, a globally defined $ = jquerry also works in webot.min.js. There is no problem so far.

However, in today's technologies, your angular react (typescript) cannot be imported into your web.min.js file directly as javascript. The main problem here is that the webot.min.js file is dependent on $, ie jerry. However, when you define webot.min.js on typescriopt, it does not open the Jquerry file that is globally here and does not work.

Problems:

**In angular**
in index.html
`<link rel='stylesheet' href='https://code.jquery.com/ui/1.12.1/themes/base/jquery-ui.css'/>
    <link type="text/css" rel="stylesheet"
          href='https://www.cyberbotics.com/jquery-ui-layout/1.4.4/layout-default.css'/>
    <link type="text/css" rel="stylesheet" href='https://www.cyberbotics.com/wwi/R2020b/wwi.css'/>
    <script src='https://code.jquery.com/jquery-3.1.1.min.js'
            integrity='sha256-hVVnYaiADRTO2PzUGmuLJr8BLUSjGIZsDYGmIJLv2b8=' crossorigin='anonymous'></script>
    <script src='https://code.jquery.com/ui/1.12.1/jquery-ui.min.js'
            integrity='sha256-VazP97ZCwtekAsvgPBSUwPFKdrwD3unUfSGVYrahUqU=' crossorigin='anonymous'></script>
    <script src='https://www.cyberbotics.com/jquery-ui-layout/1.4.4/jquery.layout.js'></script>
    <script src='https://www.cyberbotics.com/jquery-dialogextend/2.0.4/jquery.dialogextend.min.js'></script>
    <script src='https://cdn.jsdelivr.net/ace/1.2.6/min/ace.js'></script>
    <script src="https://www.cyberbotics.com/wwi/R2020b/webots.min.js"></script>`

`export class WebotsComponent implements OnInit, AfterViewInit {

  view: any;
  ipInput: any;
  portInput: any;
  connectButton: any;
  mobileDevice: any;

  constructor() {
  }

  ngAfterViewInit(): void {
    this.view = null;
    this.ipInput = null;
    this.portInput = null;
    this.connectButton = null;
    this.mobileDevice = /Android|webOS|iPhone|iPad|iPod|BlackBerry|IEMobile|Opera Mini/i.test(navigator.userAgent);
    if (this.mobileDevice) {
      let head = document.getElementsByTagName('head')[0];
      let jqueryTouch = document.createElement('script');
      jqueryTouch.setAttribute('type', 'text/javascript');
      jqueryTouch.setAttribute('src', 'https://www.cyberbotics.com/jquery-ui/1.11.4/jquery.ui.touch-punch.min.js');
      head.appendChild(jqueryTouch);

      let mobileCss = document.createElement('link');
      mobileCss.setAttribute('rel', 'stylesheet');
      mobileCss.setAttribute('type', 'text/css');
      mobileCss.setAttribute('href', 'https://www.cyberbotics.com/wwi/R2020b/wwi_mobile.css');
      head.appendChild(mobileCss);
    }

    window.addEventListener('load', this.init, false);

  }

  init() {
    this.ipInput = document.getElementById('IPInput');
    this.portInput = document.getElementById('PortInput');
    this.connectButton = document.getElementById('ConnectButton');
  }

  connect() {
    let playerDiv = document.getElementById('playerDiv');
    this.view = new webots.View(playerDiv, this.mobileDevice);
    this.view.broadcast = true;
[... 117 lines cut ...]
                    <h1>Webots streaming viewer</h1>
                    <p>
                        Connect to:ng sta
                        <input id="IPInput" type="text" value="localhost"/>
                        :
                        <input id="PortInput" type="text" value="1234"/>
                        in mode
                        <select id="mode">
                            <option value="x3d">x3d</option>
                            <option value="mjpeg">mjpeg</option>
                        </select>
                        <input id="ConnectButton" type="button" value="Connect" onClick={this.connect}/>
                    </p>
                </div>
                <div id="playerDiv"/>
            </div>
        );
    }
}

export default Webots
`
still same code but inside webot.min.js $().menu can access global  jquerry ($)

**My suggestion is that you are building this webot.min.js code with make. Embed the Jquerry library in it, so you don't need a global jquerry while webot.min.js is running. Or we can send jquery as a parameter in the constructer. In this way, not found error will not be defeated.**

omichel, 2020-09-29:
OK, I see. Thank you for the explanation.
Since we decided to get rid of jquery in webots.js, the problem will be actually solved, but in a different way.

nezihsunman, 2020-10-21:
This problem is solved for me with changing webot.min.js. Thank you so much for helping @omichel :)

closed 2020-10-21 by nezihsunman