cyberbotics/webots #2269: API for fast simulation data retrieval

Simon-Steinmann, 2020-09-25:
I did some deeper investigation and found out an issue, when trying to retrieve simulation data through the supervisor api.  I ran a c-profler on my webots controller, and I noticed that getting position data from object in the scene took very long. Here is a list of the times per call:

node.getPosition() = 23 851 ns
node.getOrientation() = 21 896 ns
Field.getSFVec3f() = 22 563 ns

If we compare that to a sensor:

PositionSensor.getValue() = 450 ns

This really becomes and issue, when needing to extract data, such as object position and orientation, or anything else from the environment. As an example, I have a reinforcement learning environment it Webots, where I have to retrieve the position and orientation of 25 objects. That is 50 calls. This takes more than 1ms, which is longer than the simulation takes for each step. 

**What is needed:**
A fast method, to retrieve all needed data.

**Ideas for API addition:**
Olivier wrote on discord:
> To speed-up things, we could implement the same mechanism as with sensors where we subscribe to a position/orientation service and get faster batched results.

I like this idea, however, perhaps it could be expanded. Perhaps useers could create a custom data service, and fill them with any Field of the scene tree. This could look something like this:
```python
timestep = supervisor.getBasicTimeStep()

field_1 = supervisor.getFromDef('box_1').getField('translation')
field_2 = supervisor.getFromDef('box_1').getField('rotation')
...
field_n = supervisor.getFromDef('box_n').getField('rotation')

dataService = supervisor.dataService.create([field_1, field_2, ..., field_n], )
dataService.enable(samplingPeriod = timestep)

data = dataService.getValue()
```

omichel, 2020-09-25:
Thank you for this nice proposal.

I have some concerns about returning a non-heterogeneous data structure (which may be difficult to handle depending on the programming language such as C/C++/Java and MATLAB). So I believe that *dataService.getValue()* should simply return an single dimension array of floating point values (double type in C) where X, Y, Z or alpha values would be added according to the order specified in the constructor.

Also from a naming perspective, *dataService* sounds too generic to me. I would replace it with *tracker*.
And why not using a normal constructor instead of a *create* method, like:
```python
tracker = supervisor.tracker([field_1, field_2, field_3])
```
Finally, if we return an array of floating point values, the method should be called *getValues()* instead of *getValue()* to be consistent with the rest of the API, see for example: https://www.cyberbotics.com/doc/reference/compass?tab-language=python#wb_compass_get_values

Would you agree on that?

omichel, 2020-09-25:
Another possibility would be that we limit the tracker to tracking only positions and orientations of objects. In that case we would pass nodes to the constructor (instead of translation and rotation fields) and the resulting array would include both the translation and rotation value for each node, e.g., 7 floating point values per node which could be easier to parse...

In most cases, I don't see anything else than translation and rotation that could useful to track this way... Do I miss something?

Simon-Steinmann, 2020-09-25:
This would definitely already be a very useful feature. I like the idea of passing nodes, and getting the poses (pos +rot) Here it would be fantastic, if we can also use` setValues()`. It could perhaps look something like this:

```python
objectPoses = supervisor.objectPoses([node_1, node_2, node_3])
poses = objectPoses.getValues()
objectPoses.setValues([pose_1, pose_2, pose_3])

I think a pose should be a float array of like this:
`[[pos], [orientation]] `
```
with orientation being the same format as the existing api, a len-9 vector for the rotation matrix.

omichel, 2020-09-28:
Yes, having a `setValues` method seems good. Regarding the naming, I would simply use "poses" instead of "objectPoses".

Simon-Steinmann, 2020-09-28:
Perhaps both variants could be provided. It could look something like this:
```
objectPoses = supervisor.objectPoses([node_1, node_2, node_3], rotFormat="matrix")
poses = objectPoses.getValues()
objectPoses.setValues([pose_1, pose_2, pose_3])
```
With `rotFormat` being one of the follwoing:
`["matrix", "axangles", "quaternion", "euler"]`

Perhaps euler should be  excluded, since it can cause problems with singularities. But in my opinion, implementing these conversions in the fast c code, would be very beneficial for users. 
Axangles are great for an intuitive human interface, but for calculations and transformations, rotation matrices and quaternions are much more powerful and easy to use.

omichel, 2020-09-28:
Yes, that's a good idea, but I would use "axis-angle" as the default value instead of "matrix".

Simon-Steinmann, 2020-09-28:
Yes, I agree. Makes the most sense. Only one more little suggestion:
instead of defining the format on construction, it could be defined on interaction:
```python
poses = supervisor.poses([node_1, node_2, node_3])

values_1 = poses.getValues("axangles") # "axangles could be omitted, since it's default
values_2 = poses.getValues("matrix")
values_3 = poses.getValues("quaternion")

poses.setValues(values_1) # Format not needed, since default
poses.setValues(values_2 ) # Format not needed, since rot shape can be identified
poses.setValues(values_3, "quaternion" ) # Format needed, since quaternion and axangles have the same shape
```

We should also decide on the formatting of a pose. Especially, whether pos and rot are returned in a single array, or whether pos and rot are seperate arrays. 
I think we have 3 options (example with 2 nodes):

**Option 1:**
```python
poses = supervisor.poses([node_1, node_2])
values = poses.getValues("axangles")
print(values)
# output would look like this:
# [[x, y, z, x_r, y_r, z_r, theta], [x, y, z, x_r, y_r, z_r, theta]]
poses.setValues(values) 
```
**Option 2:**
```python
poses = supervisor.poses([node_1, node_2])
values = poses.getValues("axangles")
print(values)
# output would look like this:
# [[[x, y, z],[ x_r, y_r, z_r, theta]], [[x, y, z], [x_r, y_r, z_r, theta]]]
poses.setValues(values) 
```
**Option 3:**
```python
poses = supervisor.poses([node_1, node_2])
pos, rot = poses.getValues("axangles")
print(pos)
print(rot)
# output would look like this:
# [[x, y, z], [x, y, z]]
# [[x_r, y_r, z_r, theta], [x_r, y_r, z_r, theta]]
poses.setValues(pos, rot) 
```

I think I would prefer Option 3. Let me know what you think :)

omichel, 2020-09-28:
First of all, I would use "axis-angle" string instead of "axangles" which seems to be a more common vocable, see https://en.wikipedia.org/wiki/Axis%E2%80%93angle_representation.
I agree, it's better to specify this rotation type argument on the getter (e.g., `getValue` method) as you propose.
About the format, option 3 seems to be good to me for Python. However in other programming languages, it should be different, like in C, `wb_supervisor_pose_get_values()` should probably return a pointer to an array of N * 7 double values.

stefaniapedrazzi, 2020-09-28:
It would also be better and more consistent with the rest of the APIs to use integers (for example class enum in C, etc.) instead of strings to specify the rotation format.

Simon-Steinmann, 2020-09-28:
We can do option 1. For matrix version, it would be a N * 12 doubles. Not too bad really.

omichel, 2020-09-28:
For C, it would be different from option 1, it would be:
```C
[x0, y0, z0, x_r0, y_r0, z_r0, theta0, x1, y1, z1, x_r1, y_r1, z_r1, theta1, ... ]
```
But I am fine with a having a different format in C and Python. So, we can go for option 3 in Python if it's the most convenient option for Python, as long as it is properly documented.

Simon-Steinmann, 2020-09-28:
C can't give you a two dimensional array?

omichel, 2020-09-28:
Yes, it can, but it's complicated as it would be a pointer to a structure containing pointers to double values. In C we usually prefer a flat data structure for this kind of arrays of floating point values.

Simon-Steinmann, 2020-09-28:
The same behavior would be nice for consistency, but the different poses should definitely be index-able. If that doesn't work in c, c should be the exception. I'm pretty sure, all the other APIs are capable of returning  vectors / arrays / lists.

Looking at this, it seems to be possible to do 2d arrays in c:
https://beginnersbook.com/2014/01/2d-arrays-in-c-example/

Simon-Steinmann, 2020-09-28:
ahh okay.

omichel, 2020-09-28:
Sure, but a C function returning a 2D array is odd and complicated, see this discussion for example: https://stackoverflow.com/questions/5201708/how-to-return-a-2d-array-from-a-function-in-c. Instead it is much more common to return a pointer to single big data buffer which contains all the floating values. Then, it's always possible to create a 2D array of pointers on the top of that, pointing to the correct values. But that's the user's responsibility.

omichel, 2020-09-29:
After think again and again about this issue, I believe we are mixing performance and programming interface which may result in a complex implementation and poor design. So, I would propose to split the problem in two: (1) performance and (2) user friendly programming interface. That should lead us to a much better result and a simpler implementation.

# 1 Performance

The main reason why the field getter functions (getSFVec3f, getPosition, getOrientation, etc.) are slow is because each getter call fires a communication loop between the libController and Webots: the libController send a request to Webots and waits for the answer. We successfully avoided this in other API functions (like DistanceSensor.getValue) by putting all the requests of a simulation step into a single buffer and getting a single answers buffer per simulation step. This is achieved thanks to the enable functions (like DistanceSensor.enable) which takes the refresh rate as an argument, so that Webots can anticipate which values will be needed at next time step. I believe we should follow the same approach to improve the performance of the supervisor field getters.

My proposal is to simply add one API function: `Field.enableTracking()` that would take the refresh rate as an argument. If tracking is enabled for a field, it will work exactly the same as with sensor data. If not enabled, it would work as before (and be slow).

I believe this is good because we could re-use the code already working for sensor data and the concept will be simple to understand for users who are already familiar with sensor data querying. And it will be fast. Another benefit is that it wouldn't be limited to translations and rotations, but would be very generic.

# 2 User Friendly Programming Interface

On the top of this low level interface, we could implement a higher level interface as proposed earlier, that could be more specific to poses (like Option 3). But this should simply be a new Python module (or a C or C++ library, etc.) that would come as a utility library, not a core libController API function. That would be something similar to the [motion API](https://cyberbotics.com/doc/reference/motion) which sits on the top of the libController API.

_Note_: the pose object could use by default the basic time step of the world file for the refresh rate of the poses. We could add a new method to the pose object to change this refresh rate if needed.

Simon-Steinmann, 2020-09-29:
I like the idea of `Field.enableTracking()`, as it would allow for fast Field values, for any field. However, I'm wondering... Giving a scenario, similar to what I described, where 50 field values have to be retrieved. Would it be faster to have a single "array of fields" call, instead of 50 single calls. A cached call takes about 400ns. If we have to extract lots of data, this can also grow. Yes, it would 'only' be 20 000 ns  for 50 calls, but perhaps this can be optimized. It is conceivable, that people use a much larger quantity of calls, where this could become an issue. I dont know how it works internally, but perhaps it is somehow possible to get the entire buffer? I have not made any tests for visual sensors, but having direct access to the buffer could have significant performance gains for RL. Would love to hear your opinions on that.

But I agree on spitting the two, and having the user friendly interface, as discussed previously, separate.

omichel, 2020-09-29:
Collecting the positions and rotations of a large number of solids will be done on the Webots side individually for each solid, then this information will be packaged in an array and sent to the libController via a pipe. The libController stores this information in structures (a data copy is performed at this point for each floating point value, but this is very fast and difficult to avoid). Then, the libController returns a pointer to these copies. So, it should be extremely fast. The only overhead I see is that you have to call the getValue method for each field. We could easily improve this if that turns out to be a bottleneck. But I would do that in a second step. To improve it, we could implement a new FieldTracker object that will somehow factorize the requests at the API level and return a bulk answer (something similar to what you initially proposed). This will minimize the number of API function calls, but honestly I am not sure it will make a big performance difference (at least in C). The biggest gain is clearly with avoiding the communication loops back and forth to Webots.

Simon-Steinmann, 2020-09-29:
That sounds good, let's do the `Field.enableTracking()` first and go from there. I will gladly analyze and take it apart, once it's done ;). Do you have an approximate ETA? It would be great (for Webots) if this fix could already be mentioned in the paper. As it stands currently, Webots would look not too great, due to the slow calls.

omichel, 2020-09-29:
Regarding the ETA, I count on you to give the answer 😉 as I understood you would implement this feature after we agreed on the design. Once implemented we will review it and merge into Webots as quickly as possible. I hope there wasn't any misunderstanding on that.

Simon-Steinmann, 2020-09-29:
I don't think I'm knowledgeable and competent enough to implement this in the webots core. I can try, but I would not count on it.

Simon-Steinmann, 2020-09-29:
[image]

Just as another visualization, this is the current c-profile of a step in our environment, retrieving the data, takes 44% of the time. 

I had a look in the source code, and I'm afraid I wont be able to implement this.

omichel, 2020-09-29:
If you study how the distance sensor works (for example) to retrieve its value from Webots, you should be able to understand how to add such a similar feature to the supervisor. The source code is located in webots/src/Controller/api/distance_sensor.c for the libController side and webots/src/webots/nodes/WbDistanceSensor.cpp for the Webots side. Then, you will like have to implement this features in webots/src/Controller/api/supervisor.c and webots/src/webots/nodes/WbSupervisor.cpp in a similar way. You probably don't need to be an expert in C/C++ for doing this. And I am happy to help you if you have any question.

omichel, 2020-09-29:
In case you don't want or can't implement it, we will probably implement it later, but I cannot tell you when as we are currently very busy with other projects.

Simon-Steinmann, 2020-09-29:
I'll have another go at it with your suggestions

omichel, 2020-09-30:
Don't hesitate to open a new PR with your very first implementation attempts and questions so that we can better help you.

lukicdarkoo, 2021-06-11:
Implemented in #2279

closed 2021-06-11 by lukicdarkoo