cyberbotics/webots #2281: Camera Image Noise

anilozdemir, 2020-09-30:
**Describe the Bug**
The simulation is adopted from Highway_overtake environment. I removed all the other cars, SUMO, etc. There is only a simple car with a front-facing camera, lots of trees around and many buildings (added from Webots objects database). The issue is the overlayed camera image includes plenty of noise (black shapes, ground texture rendered in wrong places). 

**Expected behavior**
The camera image should render the environment as is. 

**Screenshots**
[image]
[image]

**System**
 - Operating System: Ubuntu 20.04 (with i3 WM)
 - Graphics Card: two NVIDIA TITAN RTX

```
webots --sysinfo
System: Linux 5.4.0-45-generic x86_64
Processor: AMD Ryzen Threadripper 3970X 32-Core Processor
Number of cores: 64
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: TITAN RTX/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 440.100
```

**Additional context**
Added more tree to `.forest` file.

stefaniapedrazzi, 2020-09-30:
Could you please also share your project files producing the issue?

Do you have the same bug with the default `highway_overtake.wbt` world included in the Webots library?
To enable the camera device you should simply uncomment this line in the controller file:
https://github.com/cyberbotics/webots/blob/b34529dbdf110018eeb30257853b62d2bc970d88/projects/samples/robotbenchmark/highway_driving/controllers/highway_driving/highway_driving.py#L37

anilozdemir, 2020-09-30:
Please find the files in the repo: https://github.com/anilozdemir/Webots-Highway-Modification

In fact I do not have the issue when I run `highway_overtake.wbt` world. 

PS: I am also controlling the car using supervisor, by simply setting the translation coordinates. I don't know if the controller would have an impact on camera images.

stefaniapedrazzi, 2020-10-01:
The number of objects and the supervisor are not the cause of the issue.

But the are issues when merging the textures to produce the spherical camera image.
I'm investigating the problem.

stefaniapedrazzi, 2020-10-01:
I can also reproduce the issue on Ubuntu 16.04 with:
* GPU:  _NVIDIA GeForce GT 640M/PCIe/SSE2_
* driver: _4.5.0 NVIDIA 384.130_

anilozdemir, 2020-10-01:
Thanks for the update! 

Does it seem like it is fairly easy to fix? If not, what would you recommend as an alternative? (maybe not to use spherical camera?)

stefaniapedrazzi, 2020-10-01:
It is not so easy to fix.
I didn't identify exactly what is the source of the issue yet and I could reproduce it with non-spherical cameras as well.

I still don't have alternatives to completely solve it.
Disabling shadows partially works, as well as reducing the field of view.

stefaniapedrazzi, 2020-12-08:
Another workaround is to disable just the shadows of the "grass" solid, i.e. in select the `Shape` node containing the `Extrusion` and set the `castShadows` field to false.
So that shadows could be enabled for all the other objects in the scene.

stefaniapedrazzi, 2020-12-09:
This noise is due to z-fighting between the geometry and the shadows.
The best solution for z-fighting is usually to adjust the near plane on your camera as far out  as the application allows.

WREN already contains some code to prevent it but it doesn't seem sufficient in this particular case.
However, adapting it (and increasing the offset) to work in this case as well will probably lead to other unwanted effects.
https://github.com/cyberbotics/webots/blob/96b0c7fba9c1144a94e1d33d6c6722e0c7a87f21/src/wren/ShadowVolumeCaster.cpp#L350-L351

omichel, 2020-12-09:
Disabling the cast of shadows for the grass seems to be the best solution in that case, isn't it?
I don't see the interest of having the grass casting shadows. Does it cast any shadow at all?

stefaniapedrazzi, 2020-12-09:
No, there is no difference with or without shadows for the grass.
However even after disabling the shadows for the grass there are still some rendering artifacts:
[image]

[image]

-----
**Solution**
To summarize, the actions to solve all the camera rendering issues (at least on my system) are:
- disable the shadow for the grass
- set the `Camera.near` value to 0.03
- move the center of the Camera node outside the Robot (for example setting the `translation` to [0 1 1.51]).

----

I will close this issue. Please feel free to reopen it it this solution should not work.

omichel, 2020-12-09:
Thank you for this investigation.
I believe we should apply this solution to the existing simulations that we distribute.

stefaniapedrazzi, 2020-12-09:
Yes, I'm doing it.
Just note that this issue just shows up because the particular Camera settings (very wide fieldOfView set to 3.1).

closed 2020-12-09 by stefaniapedrazzi