cyberbotics/webots #2284: Play-one-frame button grays out too much

Justin-Fisher, 2020-10-01:
ARGUABLE BUG A: When you hit play-one-frame and a controller crashes, the play-one-frame button grays out and remains non-functional, even while the other play buttons remain functional and work to advance the simulation.  EXPECTED BEHAVIOR:  When the simulation is capable of advancing additional frames (without the controller that crashed) then the play-one-frame button should allow it do so one frame at a time, if you want.

DEFINITE BUG B:  If you hit Reset after the preceding, the play-one-frame button stays grayed out and non-functional.  EXPECTED BEHAVIOR: when you've reset a simulation, you should be able to play it one frame at a time, regardless of whether your preceding attempt ended with your advancing-one-frame into a controller crashing.  

ARGUABLE BUG C:  If you're running a quite slow simulation and triple-click play-one-frame, the user would expect to have the simulation advance three frames and then pause, but what it actually does is unpredictable, depending entirely upon how many frames it manages to process between successive clicks, and hence whether the button happens to be grayed out when each click occurs.  EXPECTED BEHAVIOR:  Each click of the the play-one-frame button should advance a designated pausing point by one timestep, and (if the simulation is paused) should ensure that the simulation becomes unpaused.  When the simulation reaches a designated pausing point, it should then pause.  This way three clicks of play-one-frame will always play three frames, rather than being unpredictable.

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA something fancyish

stefaniapedrazzi, 2021-03-01:
Bug B has been fixed in #2741.

"Bug A": the step button and all the other buttons works exactly the same.
* if the crashing controller is an extern controller:
    the step button remains gray and it is not possible to continue with the simulation, but the same exact behavior occurs with the real-time and run button (they can be pressed but the simulation doesn't advance)
* if the controller is started by Webots:
    the step button is enabled after the controller crash and it is possible to continue advancing the simulation step-by step

So given that the behavior is consistent, I don't see this as a bug and I think that showing that the step functionality is not enabled will help users understanding that something is wrong with the simulation and that it cannot be advanced.

About "Bug C": this doesn't seem a critical issue.
Users that want to advance the simulation step-by-step can easily check with the speedometer how many step the simulation advanced and understand that fast continuous clicking could not be processed as expected because they could occur on a disabled button.

ygoumaz, 2022-11-08:
I am closing this issue now, as the comment of @stefaniapedrazzi gives good arguments for all the issues mentioned. Feel free to reopen it if necessary.

closed 2022-11-08 by ygoumaz