cyberbotics/webots #2288: Webots crashes when ContactProperties.softCFM is 0

omichel, 2020-10-02:
**Steps to Reproduce**

- open the projects/samples/howto/worlds/cylinder_stack.wbt example.
- pause and revert the simulation.
- set the WorldInfo.ContactProperties.softCFM to 0.
- run the simulation.
- Webots crashes.

Simon-Steinmann, 2020-10-16:
Can confirm this crash. Also, it should be possible to set CFM to 0. This is the default in most simulations. As I agree on it being on by default, and it makes the simulation much more stable, there could be instances, where the user wants no error correction applied.

omichel, 2020-12-01:
Webots crashes [here](https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbSolid.cpp#L1999) as the position returned by `dBodyCopyPosition` or `dBodyGetRelPointPos` returns `{ nan, nan, nan }`.

omichel, 2020-12-01:
We already have a constraint on `WorldInfo.CFM` that should be strictly positive. I believe we should apply the same constraint to `ContactProperties.softCFM` otherwise, it causes singularities causing instabilities and generating nan values for object positions in ODE, which seems to be non-trivial to fix.

omichel, 2020-12-01:
Fixed in #2487.

PR #2487 (2020-12-01): Fixed ContactProperties.softCFM limits
Fixes #2288.
closed 2020-12-01 by omichel