cyberbotics/webots #2290: Inaccurate capsule-cylinder collision detection

lukicdarkoo, 2020-10-02:
**Describe the Bug**
Collision detection of capsule and cylinder is not accurate.

**Steps to Reproduce**
1. Open [ur10e_collisionbugwbt_caps.zip](https://github.com/cyberbotics/webots/files/5317011/ur10e_collisionbugwbt_caps.zip)
2. Rotate `wrist_1_joint: 0` to touch the cylinder

**Additional context**
Similar to #2236

Justin-Fisher, 2020-10-02:
I'm not sure whether this is the same bug, but I also get very odd cylinder-cylinder collisions, e.g. dropping a small coin-like solid onto a round tabletop.  Seems to collide further than it should, so it keeps bouncing up and dancing around for a very long time, rather than settling quickly to a stop as it does when landing on other geometries.

omichel, 2020-10-02:
Could please test the next nightly build of Webots R2020b revision 2 (to be generated tonight) and confirm whether this patch resolves also the case you observed?

lukicdarkoo, 2020-10-07:
This is also fixed in #2282

closed 2020-10-07 by lukicdarkoo