cyberbotics/webots #2291: Webots cannot run matlab

jkrichma, 2020-10-02:
Dear Webots,

Webots cannot find matlab executable on Mac OS. I have set up a symbolic link to /usr/bin/matlab  

I can run matlab from the terminal.  But, I still get this error on any matlab sample I try.

INFO: nao_matlab: Starting controller: matlab -nosplash -nodesktop -r launcher
WARNING: nao_matlab: failed to start: matlab -nosplash -nodesktop -r launcher
WARNING: Unable to find the 'matlab' executable in the current PATH. Please check your matlab installation. It should be possible to launch matlab from a terminal by typing 'matlab'. It may be necessary to add the matlab bin directory to your PATH environment variable. More information about the matlab installation is available in Webots' User guide.

Please let me know how to get around this problem. I am teaching a robotics class that uses Webots.

Thank you,

Jeff Krichmar
Department of Cognitive Sciences
2328 Social & Behavioral Sciences Gateway
University of California, Irvine
Irvine, CA 92697-5100
jkrichma@uci.edu
http://www.socsci.uci.edu/~jkrichma

omichel, 2020-10-05:
How do you start Webots? From a Terminal or from the icon?
Can you try to launch it from a Terminal to see if that works better?
Also did you set the system `PATH` environment variable to include `/usr/bin`?

jkrichma, 2020-10-05:
Hi,

I can start it from the terminal by going to the webots executable in the /Applications folder, then right clicking and so that it “Opens with terminal”. This does seem to work.  I set up a symbolic link to /usr/local/bin so that I can run Matlab from the terminal. The system PATH does point to /usr/local/bin

Thank you,

Jeff Krichmar
Department of Cognitive Sciences
2328 Social & Behavioral Sciences Gateway
University of California, Irvine
Irvine, CA 92697-5100
jkrichma@uci.edu
http://www.socsci.uci.edu/~jkrichma

TutschVaclav, 2020-11-08:
Hello, I have same problem, but if i open it from Terminal, it isnt help :( and if i use paths here: https://www.cyberbotics.com/doc/guide/using-matlab

Terminal writes: command not found

is there some way to change Webots to write in comand line: open /Applications/MATLAB_R2020b.app/bin/matlab instead of 'matlab' ? 

Thank you for your help ☺️

TutschVaclav, 2020-11-08:
Hello, I have small issue, I used sudo ln -s /Applications/MATLAB_R2020b.app/bin/matlab /usr/local/bin/matlab but webots still write:
INFO: my_controller: Starting controller: matlab -nosplash -nodesktop -r launcher
WARNING: my_controller: failed to start: matlab -nosplash -nodesktop -r launcher
WARNING: Unable to find the 'matlab' executable in the current PATH. Please check your matlab installation. It should be possible to launch matlab from a terminal by typing 'matlab'. It may be necessary to add the matlab bin directory to your PATH environment variable. More information about the matlab installation is available in Webots' User guide.

But if i write in terminal 'matlab' myself it open Matlab. I little confused what is wrong :( 

Thank you for help :)

omichel, 2020-11-09:
Did you try to launch Webots from the same Terminal as you launched MATLAB?
When launching Webots from the icon, it doesn't inherit from the standard Terminal environment.
Therefore, if you want to launch Webots from the icon, you should modify the system environment variables to add the path to the MATLAB binary in your PATH environment variable. Setting environment variables for GUI applications on macOS is however a bit tricky. See https://superuser.com/questions/476752/setting-environment-variables-in-os-x-for-gui-applications for a discussion about it.

TutschVaclav, 2020-11-10:
It started do something when i started both from terminal, but it stated write: 
INFO: my_controller: Starting controller: matlab -nosplash -nodesktop -r launcher
2020-11-10 14:31:52.100 MATLAB_maci64[54893:523857] CoreText note: Client requested name ".SFNS-Regular", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-10 14:31:52.100 MATLAB_maci64[54893:523857] CoreText note: Set a breakpoint on CTFontLogSystemFontNameRequest to debug.
2020-11-10 14:31:52.103 MATLAB_maci64[54893:523857] CoreText note: Client requested name ".SFNS-Bold", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-10 14:31:52.150 MATLAB_maci64[54893:523857] CoreText note: Client requested name ".SFNS-Regular", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-10 14:31:52.151 MATLAB_maci64[54893:523857] CoreText note: Client requested name ".SFNS-Bold", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
Error using loadlibrary
No supported compiler was found. For options, visit <a href="https://www.mathworks.com/support/compilers">https://www.mathworks.com/support/compilers</a>.
Error in loadlibrary
Error in launcher (line 86)
      loadlibrary( ...
                                                                              < M A T L A B (R) >
                                                                    Copyright 1984-2020 The MathWorks, Inc.
                                                                     R2020b (9.9.0.1467703) 64-bit (maci64)
                                                                                August 26, 2020
 
To get started, type doc.
For product information, visit www.mathworks.com.
 
Waiting up to 5 seconds for /private/var/folders/t6/x1ws4r413jqcvp30n7fhtxb80000gn/T/webots_matlab_lock to be deleted by another MATLAB instance.
Deleting /private/var/folders/t6/x1ws4r413jqcvp30n7fhtxb80000gn/T/webots_matlab_lock...
Creating: /private/var/folders/t6/x1ws4r413jqcvp30n7fhtxb80000gn/T/protofile_matlab_2020b_webots_R2020b_revision_1.m
Load Library failed.
WARNING: 'my_controller' controller exited with status: 255.

omichel, 2020-11-10:
Do you have a C/C++ compiler installed?
You probably need to install a C/C++ compiler that is supported by MATLAB.
Please refer to the MATLAB documentation about it.

lukicdarkoo, 2020-11-30:
@TutschVaclav @jkrichma have managed to make it work? I am closing the issue, but feel free to reopen it.

closed 2020-11-30 by lukicdarkoo