cyberbotics/webots #2292: insertMF causes reversion of other values

Justin-Fisher, 2020-10-02:
If you use field.setMFColor to set an entry in an LED's MF color field, the new value is immediately apparent.  But if you then use field.setMFColor(-1, newcolor) to append a new entry at the end of the MF field, this unexpectedly causes the recently set entry to to revert back to its former value.

**Steps to Reproduce**
Set up an LED with DEFname TESTLED with gradual FALSE, and the color field having the following three entries: 0 0 0, 1 0 0, 0 1 0 (i.e. black, red, green).

Run the following Python controller:
```python
from controller import Supervisor
robot = Supervisor()

colorfield = robot.getFromDef("TESTLED").getField("color")
colorfield.setMFColor(2,[0,0,1]) # overwrite green with blue
print("Before appending, color[2] =", colorfield.getMFColor(2) ) # prints blue [0, 0, 1] as expected
colorfield.insertMFColor(-1,[1,1,0]) # append yellow
print("After appending, color[2] =", colorfield.getMFColor(2) ) # prints green [0,1,0] for no good reason!
```
**Expected behavior**
The final line should still print blue [0, 0, 1] since this entry was already blue two lines up, and there's no reason that the intervening line, which simply appended yellow [1,1,0] to the end of the list, should change blue back to green.  Yellow and blue are supposed to make green in paint -- not in neighboring list elements!

It's possible that this bug is related to crash #2287 which also involved unwanted behavior when setMF and insertMF are used in the same timestep, though fortunately with color LED's we're just getting an unpredicted outcome, not the outright crash that arose in that case.

**System**
 - Operating System: Windows 10
 - Graphics Card: [e.g., NVIDIA something fancyish

Justin-Fisher, 2020-10-02:
I guess maybe the surprising part here should actually be that the first print prints blue [0, 0, 1].  Webots docs say that setMF statements will be queued and executed at the end of the time step, which fits with getting a not-yet-updated reading from the second print statement.  But it doesn't fit with the first print statement showing an updated value already.

At any rate, it looks like the expected blue value does stick in the scene tree afterwards, so the only blips have to do with the surprisingly inconsistent getMF values.  (I had initially stumbled across this in a case where I had first appended the green to the list even earlier in the timestep.  In that case, the value that stuck in the scene tree was green, which seemed wrong, but it might be partly explained if the various queued set and insert effects didn't pop off in quite the expected order or something).

stefaniapedrazzi, 2020-10-21:
This specific issue has been fixed in #2366.
As for crash #2287, the problem was indeed that the operations were not applied in the correct order.

> I guess maybe the surprising part here should actually be that the first print prints blue [0, 0, 1]. Webots docs say that setMF statements will be queued and executed at the end of the time step, [...]

The first GET is correct because the SET instructions are still stored in the controller side and the GET function retrieves the value from this list of SET instructions.
But when calling the INSERT function, the controller local list of SET instructions is sent to the Webots main application and the controller local list is cleared.
Given that the controller local list is cleared, the GET instruction doesn't find the field value locally on the controller side and has to ask Webots (where the SET instructions were applied at the end of the step and after returning the GET value).

After fixing the actions order and applying the SET instructions before the INSERT, also the second GET request retrieves the correct value.

But the issue remains if another immediate message is executed instead of the INSERT instruction (for example `supervisor.simulationResetPhysics`)

stefaniapedrazzi, 2020-10-21:
I'm inspecting sending the list of Supervisor field SET actions from the controller to Webots only at the end of the step or if related immediate messages are sent and not for any immediate message.

closed 2020-10-26 by stefaniapedrazzi