cyberbotics/webots #2294: _controller is not found when using PyCharm as in documentation

wbadry, 2020-10-03:
**Describe the Bug**
The [PyCharm](https://cyberbotics.com/doc/guide/using-your-ide?tab-language=python#pycharm) import reports an error finding `_controller`
**Steps to Reproduce**
1. Add the following to `the environment variables`

| Environment Variable 	| Location                                        	                        |
|----------------------	|-------------------------------------------------	        |
| WEBOTS_HOME          	| C:\Program Files\Webots                         	                |
| PYTHONPATH           	| C:\Program Files\Webots\lib\controller\python37 	|
| PYTHONIOENCODING  | UTF-8                                           	                                |
| Path                 	        | C:\Program Files\Webots\lib\controller          	        |
| Path                 	        | C:\Program Files\Webots\msys64\mingw64\bin      	|
| Path                 	        | C:\Program Files\Webots\msys64\mingw64\bin\cpp  	|

[image]
[image]

2. Follow [tutorial 1](https://www.cyberbotics.com/doc/guide/tutorial-1-your-first-simulation-in-webots?tab-language=python) on webots.
3. Open the `epuck_go_forward` folder in `PyCharm IDE`
[image]

4. Follow the configuration of `PyCharm` as in [webots tutorial](https://cyberbotics.com/doc/guide/using-your-ide?tab-language=python#pycharm).
[image]

5. Set controller to `extern` and run it.
[image]

6. Run PyCharm script from `PyCharm` . Even if project name is not set in environment variables, it should run the first extern controller as in documentation.

7. The script fails to find `_controller` 
[image]

**Expected behavior**
The robot should start as in tutorial. It runs well as internal webots controller.

**System**
 - Operating System: Windows 10 x64
 - Graphics Card: NVIDIA Geforce 1050 Ti

omichel, 2020-10-05:
It seems, you should also add `C:\Program Files\Webots\lib\controller\python37` to your system *Path* environment variable:
[image]
Can you confirm it works also for you like that? If so, I will fix the documentation accordingly.

wbadry, 2020-10-05:
Thanks @omichel 
I added it although it might be already added as PYTHONPATH variable as in [here](https://stackoverflow.com/questions/50278029/how-to-import-a-pyd-file-as-a-python-module)

[image]

[image]

My motivation is to take advantage of PyCharm for autocompletion of all functions and code refactoring on PyCharm. I tried to look for the issue. I don'r know if  `_controller.pyd` is used on webots text editor or something else. But it seems that even if the path is added to PYTHONPATH, path and project structure, it is not found by the compiler.

omichel, 2020-10-06:
`_controller.pyd` is located in `C:\Program Files\Webots\lib\controller\python37\`. Therefore adding it to the global `PATH` environment variable should solved the problem. It worked for me. Did you restart PyCharm after modifying the `PATH` system variable?

wbadry, 2020-10-06:
Thank you @omichel . I believe the problem is in `_controller.pyd` since the folder is seen by PyCharm. The auto-completion detected `controller.py` classes and methods which is located as you know in the same folder as `_controller.pyd`

[image]

[image]

omichel, 2020-10-06:
Since it works on my computer after adding `C:\Program Files\Webots\lib\controller\python37\` to my *Path* environment variable, I believe it was needed to be able to load `_controller.pyd` which is basically a DLL whose path should be in the *Path* environment variable to be found.

wbadry, 2020-10-06:
Do you use `Python 3.7` or calling it from `Python 3.8` on your machine?

omichel, 2020-10-06:
`Python 3.7` from https://python.org

omichel, 2020-10-06:
I found that this extra path can alternatively be set directly in PyCharm _Settings_ (from the _File_ menu):
[image]
(in my case, Webots R2020b revision 1 was installed in the `D:` drive).

wbadry, 2020-10-06:
Thanks again @omichel 

I am confident that the issue is with `_controller.pyd` and `_vehicle.pyd` not the path at all.

I created this very naive project:

- **test.py** : for just importing packages
- **helloworld.pyd** : cython dummy with hello Waleed print compiled into pyd
- **_controller.pyd** and **_vehicle.pyd** copied from webots direcotry

[image]

[image]

Without the need to add it to the path since I import it directly from code, it is not found meaning it is compiled with a different version possible a higher version that makes it unrecognized. That's why I asked if you are calling them from a project with `3.8` interpreter not `3.7`

[image]

Thanks, @omichel  for updating the issue with possible solutions.

Here is the project if you would like to quickly run it. Just check if the interpreter is set `3.7`
[test_pyd.zip](https://github.com/cyberbotics/webots/files/5334497/test_pyd.zip)

omichel, 2020-10-06:
I found yet another cleaner option. Do not set any environment variables in the system preferences, but only in PyCharm: from the _Run_ menu, select _Edit Configurations..._ and click on the icon on the right hand side of the _Environment variables:_ field. Then modify the `Path` in the _[x] Include system environment variables_ section as follow:

[image]

wbadry, 2020-10-06:
Hi @omichel 
I also tried [this step](https://intellij-support.jetbrains.com/hc/en-us/community/posts/360009758419-failed-to-load-pyd-package) which has the advantage of making the namespace come to PyCharm instead of being dimmed

[image]

I would really appreciate it Oliver if you get a chance to try the project I sent you with Python 3.7 compiler at your side to check if it works or not. If the import failed, then _controller and _vehicle need to be compiled with a 3.7 version.

omichel, 2020-10-06:
I could manage to run your project. Here is the modified version. Be sure to add `C:\Program Files\Webots\lib\controller\python37` in the PyCharm settings as shown here:
[image]
[test_pyd.zip](https://github.com/cyberbotics/webots/files/5334654/test_pyd.zip)

wbadry, 2020-10-06:
You are absolutely a magician Oliver 👍 

I cleaned up everything and this is the way that it made it work with no redundancy:

1. Just like you mentioned, adding `C:\Program Files\Webots\lib\controller\python37` in **project structure**. This is to allow `auto-completion` to work, but the code would compile even without it.
2. This is your **magic touch that made everything went to work**. The `Path` must be added by copying and pasting it and not relying on the system environment 

[image]

It is possibly a bug in PyCharm. Even identical Path added as in documents won't make it be visible unless it is added like this in environment variables:

`PYTHONUNBUFFERED=1;Path=C:\WINDOWS\system32\\;C:\WINDOWS\\;C:\Program Files\Webots\msys64\mingw64\bin\\;C:\Program Files\Webots\msys64\mingw64\bin\cpp\\;C:\Program Files\Webots\lib\controller\\;C:\Program Files\Webots\lib\controller\python37`

So, your way worked. Well done Oliver and hope to get the docs updated with this. Very awesome work.

PR #2305 (2020-10-07): Fix pycharm documentation
Fixes #2294.
Links to the new version of the documentation:
- [PyCharm](https://cyberbotics.com/doc/guide/using-your-ide?version=fix-pycharm-documentation&tab-os=windows#pycharm)
- [Extern controllers - environment variables](https://cyberbotics.com/doc/guide/running-extern-robot-controllers?version=fix-pycharm-documentation&tab-os=windows#environment-variables)
closed 2020-10-07 by omichel