cyberbotics/webots #2295: Vertical effect in spherical camera image

stefaniapedrazzi, 2020-10-05:
A user asked for vertical effect in spherical image because currently only horizontal effect is applied:

[image]
[image]

It seems correct to also expect a vertical spherical effect.
So we should maybe evaluate:
* if it is interesting to add a vertical effect
* if `spherical` is the best description for the image effect or if we could choose a better name (for example `equirectangular` or `cylindric` projection instead of `spherical` projection)

omichel, 2021-04-01:
Fix this problem would help also Robocup competitors: see https://hl.forum.robocup.org/t/modeling-wideangle-lenses-in-webots/495

omichel, 2021-05-28:
It seems this issue is not resolved, see https://hl.forum.robocup.org/t/modeling-wideangle-lenses-in-webots/495/7

stefaniapedrazzi, 2021-05-31:
Related to #3068

stefaniapedrazzi, 2021-06-07:
To get both vertical and horizontal effect we should use the stereographic projection instead of the equirectangular projection (currently implemented).

TheMangalex, 2022-09-16:
I'm not sure if this issue is still being worked on, or if there is some finished implementation on the develop already, but in case not, I made my own version of the shader which makes a distortion in both directions, but I'm currently not sure, if it is correctly scaled. (I used a reference implementation as basis and have to recheck the math).
[image]

Also it seems that replacing the merge_spherical shader is not trivial, as it is also used for the LIDAR projections. Therefore different handling of equirectangular and fisheye projections would be needed.

If you are interested in this fisheye shader, I can share it with you.

omichel, 2022-09-16:
Thank you @TheMangalex. It looks very promizing. Could you please open a new PR with your contribution? It should target the develop branch. We will review it in depth in order to merge it in the develop branch as soon as possible.

stefaniapedrazzi, 2022-10-25:
Fixed in #5240 and #5266.

PR #2961 (2021-04-14): Fix Camera lens distortion
Fix #2295 and #2934:
Camera lens distortion uses the Brown's distortion model, but there were bugs in the implementation and the center was not used for the radial and tagential components.
Additional, to apply lens distortion it is needed to apply an inverted Brown's distortion model.

Now, the distortion is correctly applied both in X and Y directions, as well as symmetrically on the right/left of the image. Here is the result using `Lens { center = [0.5, 0.5], radialCoefficients = [-0.5, 0], tangetialCoefficients = [0, 0]}`:

[image]
closed 2022-10-25 by stefaniapedrazzi