cyberbotics/webots #2313: Pointclouds generated from Lidar nodes with non-zero roll are skewed.

dscroft, 2020-10-09:
**Describe the Bug**
A clear and concise description of what the bug is.
Pointclouds generated by Lidar nodes with non-zero roll appears to have an additional skew applied to them. When visualised in Webots the point cloud can be seen not aligning to objects issue does appear to be in the underlying point cloud values and not just a rendering error in Webots as the additional skew can be seen in the point cloud if it is visualised in other software (i.e. pcl).

**Steps to Reproduce**
Git repo here -> [https://github.com/dscroft/webots_bug_report](https://github.com/dscroft/webots_bug_report)
contains world and controller code for visualisation of the point cloud outside of Webots (requirements pcl).
1. Open gimbal.wbt
2. Set roll motor position to not 0
3. Turn on point cloud rendering

Screenshot here -> [https://raw.githubusercontent.com/dscroft/webots_bug_report/main/images/wall_skew_webots_and_pcl.png](https://raw.githubusercontent.com/dscroft/webots_bug_report/main/images/wall_skew_webots_and_pcl.png)
Shows pointcloud in both Webots and pcl

**Expected behaviour**
Pointcloud point positions to match object positions

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Linux Mint 20
 - Graphics Card: NVIDIA RTX 2070

DavidMansolino, 2020-10-09:
It seems there is indeed something wrong with the projection:
[image]
The top line should be straight.

This is easy to reproduce with your world file even with the ``void`` controller and enabling the point-cloud mode in the robot-window.

DavidMansolino, 2020-10-09:
Note that setting the ``near`` field to the default value (0.01) works better, but still there is a distortion.

ruca03, 2026-10-01:
Hi, 

Is there any updates about this bug?