cyberbotics/webots #2323: Error after adding battery on DJI Mavic2Pro drone

kosnet2, 2020-10-12:
- I am gettting this warning after adding a battery on the existing dji Mavic2Pro drone with predefined values

```
  field MFFloat    battery         [50, 100, 30]
  field SFFloat    cpuConsumption  10
```
`WARNING: Mavic2Pro (PROTO) > DEF FRONT_RIGHT_HP_HELIX Propeller > RotationalMotor: Feedback is available for motorized joints only`

- If I don't add the battery the controller is giving this error:

`WARNING: Mavic2Pro (PROTO): 'wb_robot_battery_sensor_enable' called while the 'battery' field is empty.`

- If go with the first solution but leave the battery fields empty (ie)

```
  field MFFloat    battery         []
  field SFFloat    cpuConsumption  10
```
and then add the battery values through the webots hierarchy editor, Webots **crashes**

Does anyone have a clue what am I doing wrong ?
My controller is in python

DavidMansolino, 2020-10-13:
Which version of Webots are you using?
We fixed several issues related to battery recently (which sounds like what you are mentioning):
> Fixed crash when calling the wb_robot_battery_sensor_enable() function if the battery field of the Robot is empty (#1801).
>Fixed annoying warning messages when calling the wb_robot_battery_sensor_enable() function if the Robot contains a Track node (#1801).

But it looks indeed that the 'annoying warning messages' are still present if the Robot contains a Propeller node. We will fix this soon. In the meantime, you can use the first option and ignore the warning messages.

kosnet2, 2020-10-13:
@DavidMansolino Thank you for your feedback :)

I am currently using Webots R2020b revision 1 and I have installed the latest update.

Is there any way to switch off the warnings ?

DavidMansolino, 2020-10-13:
To 'hide' them you hve to right click on the Webots console and select all the filters except 'Others'.
Note also that I just fixed the issue in  #2324. The fix will be included in the nightly build of R2020b-rev2 from tomorrow: https://github.com/cyberbotics/webots/releases

closed 2020-10-13 by DavidMansolino