cyberbotics/webots #2338: Python's Robot.getDevice() crashes

Justin-Fisher, 2020-10-14:
The Python API includes a definition for `Robot.getDevice` despite this being undocumented.  The Webots docs encourage you to use more specific functions like `Robot.getGPS` instead, to ensure that the device you get will be a member of the appropriate subclass.  However, when I look at the actual source code for Webots `controller.py`, the function `__getOrCreateDevice` automatically detects each device's type and generates a device of the appropriate subclass, and this code seems to be used by each of the specific functions like getGPS, so I don't think the more specific call should really be necessary.  This is evidenced by the fact that Python's generic `getDeviceByIndex` works fine without needing different versions for the different types of devices.  So there's no reason `getDevice` couldn't work too.

Unfortunately, the existing Python `def getDevice(tag)` function suffers several problems.  It is defined as a `Robot` class method, but is missing the initial `self` argument that is required for such methods in Python, so it always crashes.  ( `getDevice( x )` has one too many arguments, and `getDevice()` treats the robot itself as the `tag` argument, which of course doesn't work.)  Furthermore, the given argument is called `tag` but Webots docs discuss the relevant arguments to various Python device-getting functions as being "names" or "indexes", not "tags" (which seem to be an internally used label for something that behaves much like an index).  The code itself presumes that the given `tag` must be a number, and will crash otherwise when it compares `tag` to the length of its list of devices.  This code could handle `robot.getDevice(1)` but not `robot.getDevice("gps")`.  This is odd when the analogous `get_device` functions in C and Matlab take a _string_ rather than an integer as their argument, and there already is another Python function `getDeviceByIndex` (with analogs in C and Matlab) which does look up devices by index.

**Proposed bugfixes:**  (1) Python's `Robot.getDevice` should include the required `self` parameter, becoming `getDevice(self, name_or_index)` so that it won't always crash. (2) Python's `getDevice` should accept string arguments, to parallel the corresponding functions in C and MatLab, and to distinguish it from `getDeviceByIndex`.  (3) Python does not require that an argument be of a pre-specified type, so `getDevice` could still accept integer index arguments too, which strikes me as useful, and better than making coders try to remember whether you called the indexed version "FromIndex" or "ByIndex" (the source code uses both of these, at different levels!).  (4) Documentation for Python's version of `Robot.getDevice` should be added matching whatever functionality it ends up having.  (5) The documentation encouraging using specific getX functions rather than this more generic `getDevice()` should be removed.  (In the source code, the only other thing the specific getX functions do is check that the device with that name really is a Webots device of the specified type, so e.g. `robot.getMotor("gps")` will return `None` if the device named "gps" isn't a motor.)

**Suggested Code.**  Note, the first line adds the missing `self` argument required in Python class methods, and makes clear that the given argument can be a device name or index number.  The `if` condition detects string inputs and treats them in exactly the same way names are treated when passed to specific commands like `getGPS`, whereas the `else` condition instead treats the parameter as an index number and treats it in exactly the same way as does `getDeviceByIndex` which is documented to work and has less redundancy than the original code with what is done by the final line.  The final line then calls a function that gets/creates the relevant device, doing so in a way that ensures that a device of the relevant subclass will be created.   

```python
    def getDevice(self, name_or_index):
      """This returns this robot's device with the given name if the given argument is a string,
      or with the given index number otherwise.  Returns None if there is no such device."""       
      if isinstance(name_or_index,str):
        tag = self.__internalGetDeviceTagFromName(name_or_index)
      else:
        tag = self.__internalGetDeviceTagFromIndex(name_or_index)
      return self.__getOrCreateDevice(tag)
```

DavidMansolino, 2020-10-15:
Just for the note, this is an internal function that we use indeed for creating the other get_DevicenName_ functions, this is why it is not very intuitive and not documented/discouraged to use.
But it would be indeed nice to improve it like you suggest so that user can use it, this does not concern Python but all the OO languages.

omichel, 2020-10-15:
As a consequence of this, the C++ and Java API should also provide the generic `getDevice()` method. However, in these languages, I would probably keep the specific methods (e.g., `getGPS()`) as the preferred method because it is not possible to return different class types from a method. Therefore the generic `getDevice()` method would return an object of the `Device` base class, as with `getDeviceByIndex()` which is less convenient to handle than the specific methods as it would require the follow C++ code:

```C++
Camera *camera = dynamic_cast<Camera *>(robot.getDevice("camera"));
```

Which looks not as elegant as:

```C++
Camera *camera = getCamera("camera");
```

To keep also some consistency between C++ and Python and avoid breaking backwards compatibility, we should keep the specific methods in Python, in addition to the generic one which is arguably more convenient to use and more consistent with the C API, and therefore should be documented as the preferred method.

omichel, 2020-10-15:
I just implemented this idea [here](#2341) and it seems to work nicely. @Justin-Fisher: would you agree to contribute to the documentation on this PR? I just sent you an invite to become a committer on this repository.

omichel, 2020-12-07:
Fixed in #2510.

PR #2510 (2020-12-07): Improved device getters in the Python API
Resolves #2338.

Remains to be done:
- [x] update all the python examples to use the new `getDevice()` method.
- [x] [documentation](https://cyberbotics.com/doc/reference/robot?version=feature-improve-python-device-getters&tab-language=python#wb_robot_get_device).
closed 2020-12-07 by omichel