cyberbotics/webots #2342: Move the object oriented API sources to a better folder

omichel, 2020-10-15:
Currently the source code for the object oriented APIs (C++, Python, Java and MATLAB) is located in [webots/resources/languages/](https://github.com/cyberbotics/webots/tree/master/resources/languages) which seems odd to me.
I believe this is inherited from the old time when Webots was not open source and we wanted to distribute the source code of these APIs, but we didn't want to distribute anything from the [src](https://github.com/cyberbotics/webots/tree/master/src) folder.
I think it would be at a better place in [webots/src/languages](https://github.com/cyberbotics/webots/tree/master/src/languages). @cyberbotics/maintainers: do you agree with that?

stefaniapedrazzi, 2020-10-15:
why not creating different folders for each languages in `webots/src/Controller` (`webots/src/Controller/c/`, (`webots/src/Controller/cpp/`, etc) ?
This would be a more intuitive structure for me.

In any case we have languages files in many different locations:
the Matlab API generation code is in the `webots/lib` folder (and ROS API in the `webots/projects/default`).

omichel, 2020-12-04:
Fixed in #2494.

closed 2020-12-04 by omichel