cyberbotics/webots #2343: Fix human animations in latest version of Webots

stefaniapedrazzi, 2020-10-15:
The simulation of animated human with `Skin` node is no longer correctly working in the latest version of Webots.

I just checked again and for example in Webots R2019a revision 1 they are working correctly.

stefaniapedrazzi, 2020-11-04:
It seems that the bone index changed and the ones used in the controller and retrieved from the BVH animation file don't match any longer the ones read by Webots from the FBX.

stefaniapedrazzi, 2020-11-04:
The Webots internal bones order changed due the libassimp update.
Given that root bone index was hard coded in the `bvh_animation.c` controller, the wrong bone was moved.
Fixing the `bvh_animation.c` so that it retrieves the root bone index based on the name fixes the issue and the same controller works correctly both in Webots R2020b and Webots R2020a.

closed 2020-11-04 by stefaniapedrazzi

omichel, 2020-11-05:
So why did you close this issue? Is `bvh_animation.c` already fixed?

stefaniapedrazzi, 2020-11-05:
The human animation sample simulation and the `bvh_animation` controller is not available on this webots repo yet.
I will port it from the old repository when addressing #2345.