cyberbotics/webots #2352: Updating a label doesn't always change it

Justin-Fisher, 2020-10-17:
Webots labels (the text displays that supervisors can print atop the simulation) don't change when you pass the same text string again, even if you pass a new color specification.  If you want to change a label's color without changing its text, you currently need to send two setLabel() commands, perhaps even during the same timestep: one to change the text to a new value, then another to change the text back to its old value. 

Expected behavior would be that any change to the label would show up in the display, not just changes that include changes to the text string.  

My guess is that some sort of checking is done to avoid expending effort in cases where a setLabel() command is sent that wouldn't actually change the label at all.  However, this currently ends up being too lazy, refusing to expend effort when an important change about the label, like color, actually has been requested.  I haven't tested other label features, like size or location, to see whether they also require a change in text to implement.

omichel, 2020-10-19:
I just tested it: unlike changing the color only, changing the size only or location only of a label does not cause the bug reported here.

closed 2020-10-19 by omichel

Justin-Fisher, 2020-10-19:
Thanks!  That'll let me remove the jerry-rigged workaround I'd been using, sending two commands in a row to change a label's color.

PR #2357 (2020-10-19): Fixed label color-only change
Fixes  #2352.