cyberbotics/webots #2374: Deploying behavior and controlling Nao in Dockerized Webot

nezihsunman, 2020-10-21:
We have installed the Webots container you use on Docker. We managed to run a virtual nao robot following this link: https://cyberbotics.com/doc/guide/web-simulation. 
 
In our own project, we want to use AWS and KUBERNATES technologies as a structure that works over the web project and communicates with the webots docker container.
 
Architecturally, we want to connect a docker container with the "Django(Python language)" as the port running on webots. In frontend, we can access to the real time view of the Nao. But there are several issues with controlling Nao. We have some questions regarding to this topic:
 
1) Is it possible to deploy custom behavior packages to Nao that is simulated by Webots?

 2) If it is possible how can we call and use deployed Nao behaviors?
 
Previously, we were using pynaoqi and Choregraphe to deploy and use custom behavior packages. For example, we used the following code to connect to the Nao and call StandUp behavior.
 
robotIP = "127.0.0.1"

managerProxy = ALProxy("ALBehaviorManager", robotIP, 9559)

managerProxy.runBehavior("standup-a650fd/StandUp")

omichel, 2020-10-22:
To deploy custom behavior packages, you have several options:
1. Create a custom Docker image (inheriting from ours, and adding your custom worlds and controller files).
2. Execute a script that will download and install (from GitHub or any other location) your custom worlds and controller files.

_Note_: we had very poor experience with naoqi support for simulation and decided to drop it. Instead, it is much more reliable to use the native Webots API to control the NAO robot, either in Python or C/C++.

nezihsunman, 2020-10-22:
We currently have a nao robot behavior that exists, but this structure functions with the nao robot that works locally and is physically present. We are trying to bring this structure to the online process due to the current corona situation.

The other question here is are protocols written to communicate with the webot docker in terms of nao behaviors? If so, test environments and at what level? Have these protocols been tested on any nao robots?

As far as I know from the documentation, JavaScript and php based structure is currently working. There are some python APIs.

Apart from these, are there any communication protocols that have been tried in the development process or can be made in the future?

umutcakan, 2020-10-22:
I know the experience with naoqi you had before, I have tried to use naoqisim. I have checked the native Webots API, especially **nao_demo_python**, but I found related information regarding the sensors and motion control. 

Unfortunately, I could not see any information regarding deploying & using behavior.  Our problem is that currently, we have no clue about how to perform deploy & use a custom behavior package to virtual nao powered by Webots. 

I will appreciate if you can share some code snippet or an example about using behavior packages. It is also okay if we can access the built-in behavior packages (e.g. StandUp) for the beginning.

Thanks for your help,
Best regards.

omichel, 2020-10-23:
What do you mean by "deploying & using behavior"? Transferring the control program to a real NAO robot?

umutcakan, 2020-10-23:
Behavior is a set of motion defined by Choregraphe. For example, instead of writing control codes for motor by motor, it contains a set of control commands that enable Nao to move accordingly. For instance, the Nao robot has a built-in 'behavior package' called StandUp. When you run 'use' this behavior, it basically makes Nao stand up. 

This repo contains the aforementioned behavior package: https://github.com/umutcakan/Nao-StandUp-Behavior

Thanks for your help.

omichel, 2020-10-23:
OK, I see, in that case, the concept of behavior is very similar to the concept of motion we have in Webots. If you run the nao_demo.wbt sample world and look into the source code of the controller, you will see that it loads and executes different motion files, like walk forward, side step to the left, turn 180 degrees, etc. These motion files are located here: https://github.com/cyberbotics/webots/tree/master/projects/robots/softbank/nao/motions
However, converting the existing NAO behavior files to Webots motion files is not straightforward, but certainly doable.

nezihsunman, 2020-10-23:
If we want to integrate the behavior system to NAO, will both TCP / IP or other type of protocol from outside allow it?

Will the streaming server work together to visualize to "Frontend JavaScript" and control from python backend with the TCP / IP control mechanism or other controls?

Which system do you recommend if system want to be controlled by python from the outside?

Thanks for your help 💯

omichel, 2020-10-23:
Yes, that should work. There are plenty of libraries in Python that should allow you to communicate over TCP/IP, I don't have any special recommendation for that.

lukicdarkoo, 2020-11-30:
I think we should close this one. Feel free to reopen if you have any other related questions.

closed 2020-11-30 by lukicdarkoo

umutcakan, 2021-03-11:
Hello, Olivier again! I have managed to create a converter that transforms behavior packages into motion files. But in some of our behavior packages, there are 'hand' movements, like waving, etc. I could not manage to control Nao's hands with a motion file. I also checked the actuator names, but there is nothing regarding Nao's hands. There are 2 different hand actuator fields in the behavior package called 'RHand' and 'LHand'. 

I will appreciate it if you can share some information about this. Thanks for your help.

omichel, 2021-03-11:
You need to enable 25 degrees of freedom in the [Nao](https://github.com/cyberbotics/webots/blob/master/projects/robots/softbank/nao/protos/Nao.proto) robot (which should be the default).
Then, the finger names can be found in the [NaoRightWristH25Realistic.proto](https://github.com/cyberbotics/webots/blob/master/projects/robots/softbank/nao/protos/NaoRightWristH25Realistic.proto) and [NaoLeftWristH25Realistic.proto](https://github.com/cyberbotics/webots/blob/master/projects/robots/softbank/nao/protos/NaoLeftWristH25Realistic.proto) proto files.

umutcakan, 2021-03-11:
Thanks for the quick response Olivier. 25 degrees of freedom was set to default as you said. But I quite did not understand about calling fingers, is it the same procedure that we have done so far? Like filling angles in the motion file (e.g. HeadPitch, HeadYaw, etc.) or should I call them by calling LWristYaw, etc.?

omichel, 2021-03-11:
Yes, it's the same, you should simply use the finger joint names, e.g., LWristYaw, LPhalanx1, LPhalanx2, LPhalanx3, LPhalanx4, etc.

omichel, 2021-03-16:
You need to call:
```python
self.getDevice("LPhalanx1").getPositionSensor().enable(16)
self.step(16)
self.getDevice("LPhalanx1").getPositionSensor().getValue()
```

umutcakan, 2021-04-17:
Hey Olivier!

Thanks so much for your help! I've managed to create a proper controller from the example python controller that you have provided. I could convert custom nao behaviors into motion files. At this point, I have 2 issues:

1) I inspected the controller from this link: https://cyberbotics.com/doc/guide/controller-programming. What I am trying to do is that I have many motion files and I need to call a specific motion file at a time and after finishing playing it, notify the controller. For example, I have preloaded a, b, c motion files, I need to call motion a file, and when it is finished (the motion), I need to return a message or some sort of output to the main controller which is used for other purposes. For this purpose, I wanted to ask you if there is an API call that notifies you when a motion file is executed and finished when we run it? For example:

self.testMotion.setLoop(False)
signal_that_indicates_motion_is_finished = self.testMotion.play()

Or is there a way that I can use with basicTimeStep maybe? Currently, I am looping infinitely with the loop that you provided in the example code:

while robot.step(self.timeStep) != -1:

2) We have dockerized webots server on localhost currently, taken this link as a guideline: https://cyberbotics.com/doc/guide/web-simulation#using-docker. Currently, the controller works fine with the desktop webots app, I set all the environment variables, etc. What I am trying to do here is that, converting the aforementioned controller to a web controller that will play motions on a web simulation.  But unfortunately, I could not find a proper way to do this. I was wondering if there is an example controller/guideline to fulfill this purpose?

omichel, 2021-04-19:
1. You can use [self.testMotion.isOver()](https://cyberbotics.com/doc/reference/motion?tab-language=python#wbu_motion_is_over) to determine if a motion is finished or even anticipate it by getting its duration with [self.testMotion.getDuration()](https://cyberbotics.com/doc/reference/motion?tab-language=python#wbu_motion_get_duration).
2. We provide a guide to deploy simulations on the web [here](https://github.com/cyberbotics/webots-cloud/wiki).

umutcakan, 2021-04-19:
Thank you Olivier for your response !

isOver method is exactly what I needed for my work. Getting duration is also a nice method that I am willing to use for different purposes!

For my 2nd question, we have managed to deploy simulation on a web server. But with my current controller, it's not possible to control a deployed simulation, it works only on desktop app. I was wondering if there is an example of a controller (like the python one) to control deployed simulations. For example, a python controller that controls the web simulated nao or other robots etc.