cyberbotics/webots #2379: import _controller ImportError: DLL load failed

LIU-yx97, 2020-10-22:
> I believe I found the problem: you should add an extra PATH to your environment variables:
> `PYTHONUNBUFFERED=1;PATH=G:\Program Files\Webots\lib\controller\;G:\Program Files\Webots\msys64\mingw64\bin`
> 
> Can you try that and confirm it works?

Yes!!! I have tried your solution and it works very well!!! Thank you very much for your detailed help!

_Originally posted by @Weiyi-Zhang258 in https://github.com/cyberbotics/webots/issues/1399#issuecomment-589904387_

I also encountered the same problem and modified it in the way described above, but the problem still existed.
This is my Environment variables:
PYTHONUNBUFFERED=1;
PATH=E:\Program Files1\webots\Webots\lib\controller\;E:\Program Files1\webots\Webots\msys64\mingw64\bin\;E:\Program Files1\webots\Webots\msys64\mingw64\bin\cpp\;;
PYTHONPATH=E:\Program Files1\webots\Webots\lib\controller\python37

LIU-yx97, 2020-10-22:
@omichel I would be most grateful if you could help me!

omichel, 2020-10-22:
The anti-slash after `controller\` and after `bin\` is missing in your PATH. This is critical. Please refer to https://cyberbotics.com/doc/guide/using-your-ide?tab-os=windows#pycharm where the anti-slashes are part of the PATH set in PyCharm. I guess this is probably a bug of PyCharm...

closed 2020-10-22 by omichel

huiku2021, 2022-05-27:
I followed the settings in the documentation, but I still have this problem.

omichel, 2022-05-30:
> I followed the settings in the documentation, but I still have this problem.

Which version of Webots are you using? On which operating system?

yasinsahn, 2022-06-01:
I have the same error with Windows11, Python 3.9 and Webots R2022.a
PATH=C:\Users\yasin\AppData\Local\Programs\Webots\lib\controller\;C:\Users\yasin\AppData\Local\Programs\Webots\msys64\mingw64\bin\;C:\Users\yasin\AppData\Local\Programs\Webots\msys64\mingw64\bin\cpp;
PYTHONIOENCODING=UTF-8;
PYTHONPATH=C:\Users\yasin\AppData\Local\Programs\Webots\lib\controller\python39;
PYTHONUNBUFFERED=1;
WEBOTS_HOME=C:\Users\yasin\AppData\Local\Programs\Webots

huiku2021, 2022-06-02:
> I have the same error with Windows11, Python 3.9 and Webots R2022.a
> 
> PATH=C:\Users\yasin\AppData\Local\Programs\Webots\lib\controller\;C:\Users\yasin\AppData\Local\Programs\Webots\msys64\mingw64\bin\;C:\Users\yasin\AppData\Local\Programs\Webots\msys64\mingw64\bin\cpp; PYTHONIOENCODING=UTF-8; PYTHONPATH=C:\Users\yasin\AppData\Local\Programs\Webots\lib\controller\python39; PYTHONUNBUFFERED=1; WEBOTS_HOME=C:\Users\yasin\AppData\Local\Programs\Webots
> 
> I would be very happy if you could help me. Thanks in advance!

hello！I have solved the problem.win10,python3.8,webots r2022.a.Environmet variables:
PYTHONUNBUFFERED=1;PYTHONIOENCODING=UTF-8;PYTHONONPATH=E:\webots\Webots\lib\controller\python38;Path=E:\webots\Webots\lib\controller\;E:\webots\Webots\msys64\mingw64\bin\;E:\webots\Webots\msys64\mingw64\bin\cpp
You can refer to this.Good luck to u!

huiku2021, 2022-06-02:
> > I followed the settings in the documentation, but I still have this problem.
> 
> Which version of Webots are you using? On which operating system?

thank you, i have solved the problem.

ibnukoel, 2022-06-23:
Awesome, I did it, I can solve the same problem, after I installed python 3.7.8 in my pc. 
I used python 3.8 and it didn't work to use controller.dll with pycharm. 

thanks everyone

Corsair-cxs, 2023-07-15:
> hello！I have solved the problem.win10,python3.8,webots r2022.a.Environmet variables: [...]

Thank you! I solved the same problem on webots 2023.a

PR #2380 (2020-10-22): PyCharm documentation
Added some precision.
Fix #2379.