cyberbotics/webots #2384: Joint position tolerance issue in trajectory_follower.py

s-trinh, 2020-10-23:
Hi,

I am using the `trajectory_follower.py` as reference code since it is useful for my work. 
The following code looks not ok:

https://github.com/cyberbotics/webots/blob/f7809e1f2e54a64d19c99bbc0c282159b3ae0db1/projects/robots/universal_robots/resources/ros_package/ur_e_webots/scripts/trajectory_follower.py#L208-L222

Check for joint positions convergence is calculated at L212 for the final trajectory point using `self.joint_goal_tolerances` threshold.
But the variable is overwritten at L218, L219 using hard-coded thresholds.
Also, current joint positions are retrieved at L211 and `self.motors[i].setPosition()` is called at L215. Thus, when the joint positions are checked, they seems to be done not exactly on the current joint positions.

`setpoint = sample_trajectory()` at L213 seems not needed since we should want to just apply the final trajectory point, and we have `last_point = self.trajectory.points[-1]` done at L210.

Finally, default joint position tolerances seems for me too high:

https://github.com/cyberbotics/webots/blob/f7809e1f2e54a64d19c99bbc0c282159b3ae0db1/projects/robots/universal_robots/resources/ros_package/ur_e_webots/scripts/trajectory_follower.py#L130

It should be the controller for the URe, so a tolerance of `0.05 rad` would mean that we would consider the arm has reached its final position when the joint angle errors are below ~2.86deg (moreover it is actually ~5.73 deg with 0.1 value used at L218)? This seems for me too high thresholds.

omichel, 2020-10-26:
Are you getting better results with lower thresholds?

s-trinh, 2020-10-26:
Sorry, I did not have performed any comparison.

Now I am confused. Does this line already send the last trajectory point?

https://github.com/cyberbotics/webots/blob/f7809e1f2e54a64d19c99bbc0c282159b3ae0db1/projects/robots/universal_robots/resources/ros_package/ur_e_webots/scripts/trajectory_follower.py#L202

---

Thinking about it, in most case since the trajectory is interpolated, the current robot joint positions just before sending the last trajectory point should be really close to the final desired joint positions. So probably the joint position errors should be already below the joint goal tolerances. I believe also that `self.motors[i].setPosition(setpoint.positions[i])` will perform the PID control to reach the target joint positions. But I think that in theory it can happen that `goal_handle.set_succeeded()` is called but the arm is still moving due to the PID control of `self.motors[i].setPosition(setpoint.positions[i])`?

---

So what is the reason to use `[0.1] * 6` and `[0.05] * 6` as position and velocity tolerances?

DavidMansolino, 2020-10-27:
> Now I am confused. Does this line already send the last trajectory point?
> 
> https://github.com/cyberbotics/webots/blob/f7809e1f2e54a64d19c99bbc0c282159b3ae0db1/projects/robots/universal_robots/resources/ros_package/ur_e_webots/scripts/trajectory_follower.py#L202

The last trajectory point is sent in the next condition: https://github.com/cyberbotics/webots/blob/f7809e1f2e54a64d19c99bbc0c282159b3ae0db1/projects/robots/universal_robots/resources/ros_package/ur_e_webots/scripts/trajectory_follower.py#L208

> Thinking about it, in most case since the trajectory is interpolated, the current robot joint positions just before sending the last trajectory point should be really close to the final desired [...]

Yes that's indeed totally true!

> So what is the reason to use `[0.1] * 6` and `[0.05] * 6` as position and velocity tolerances?

These values were chosen to be the exact same as the ones used by the ROS driver of the real UR arms:
https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_driver/src/ur_driver/driver.py#L816-L817

s-trinh, 2020-10-27:
Thanks for your answers. Feel free to close the issue.

I think the code can be made clearer with the different points I mentioned in my first post. But ultimately, this would not change the robot behavior.

omichel, 2020-10-27:
Thank you.
Feel free to open a PR to add comments or clarifications in the code if you believe it could be useful.

closed 2020-10-27 by omichel