cyberbotics/webots #2385: Type annotations for Python API

PeterJCLaw, 2020-10-25:
I'm not sure how the Python API is built, however I was wondering whether it would be possible to create type annotations for it. This would help document the APIs as well as allow users to check that their code is using the API correctly (just as, for example, the Java compiler is able to do for users of the Java API).

At the moment I've got a set of [local stubs in my project](https://github.com/srobo/competition-simulator/tree/master/stubs) which I've manually created based on the documented API, with the types captured from a mixture of observed data types and the docs for other languages (mostly Java).

It feels like type annotations could usefully be added to the signature snippets in the docs as well as potentially being published as a stubs package on PyPI for use in type checking and as editor completions.

I'm definitely hoping that the signatures would be generated from the true source, however I'd be happy to contribute the stubs I've created as a starting point if they need to be manually defined.

omichel, 2020-10-26:
We use swig to generate the Python API from the C++ API and there is no easy way to introduce type annotations there, see https://github.com/swig/swig/issues/735 for a discussion on this topic. So, you local stubs approach may be a better way. I guess we shall it to the Python library build process somewhere here: https://github.com/cyberbotics/webots/tree/master/resources/languages/python
Feel free to propose a solution by submitting a new pull request in this repository. We will be happy to review it.

Alternatively, it shouldn't be that difficult to get rid of SWIG for the Python wrapper and maintain it, based on the C libController library (instead of the C++ one). But that's probably a bit more of work, but could be also very nice and fix other problems caused by SWIG.

omichel, 2021-10-13:
All what you propose should be easy to implement if we totally get rid of swig for generating the Python API.
I believe more and more that a totally new Python API could be created without swig and maintained manually, which should be easy because it will be very straightforward. This API could rely on the C API (instead of the C++ one) thus, removing the dependency on `libCppController` and other C++ dependencies. It would call the C API function directly from python.
Such a library could easily include python hints and documentation with hyperlinks as described above.
In addition to all these benefits, such a library should be designed to be more "pythonic" with python `.value` getters and setters instead of `.getValue()/.setValue()` methods inherited from C++.

If someone would volunteer to go ahead with such an implementation, I would suggest to start with a minimal proof of concept implementation supporting only one type of sensor and one type of actuator. I would be happy to review it in a PR and if it looks good we could generalize it to the complete API. Ultimately, we could drop the current Python API and replace it with this new one.

omichel, 2021-10-15:
Your type hinting suggestion sounds very good to me.
About adding the `interval` optional argument, this is also a very good idea. However, it may be a problem with some devices, such as `Emitter`, `Pen`, `Skin` or `LED`, which don't have a `enable` method, but I guess in this case, the library should simply issue a warning in the console.

PeterJCLaw, 2021-10-15:
I'd also encourage having separate per-device-type methods, or at least supporting that as part of the API. Our project currently makes use of its own wrapper to essentially achieve that: https://github.com/srobo/competition-simulator/blob/a49f5c3fa51ee35458b4194d5d0fae4810ddeb14/modules/sr/robot/utils.py#L30-L57. This approach works, but isn't especially nice.

As you noted, having true per-device-type methods opens up the API to doing other things with the argument which can then be customised per the type of device.

Justin-Fisher, 2021-10-15:
Before embarking on this, I should probably try to get clear on something else you said above, suggesting that we might want to use the C API instead of the C++ API that swig had been using.  I had been thinking of starting with a current working version of controller.py and changing it in ways that preserve all of the underlying functionality, while adding various bells and whistles like type-hinting, documentation strings, new methods, and handling of additional/optional arguments.  I think this approach would most naturally end up still using whatever controller.py currently uses, which I think is the C++ API.  What would be the advantage of using the C API instead?  How hard is it to translate between these?

Justin-Fisher, 2021-10-16:
Another relevant question is what version(s) of Python we'll be aiming to support?  There's a fairly deep division in the Python community between people who use a quite new version of Python 3, and old sticks in the mud who have stuck with Python 2 for 13+ years since Python 3 wasn't entirely backwards-compatible with Python 2.  I believe that Webots has been using swig to create separate versions of controller.py for several recent versions of Python3 (3.7, 3.8, and 3.9), and for an ancient version of Python 2.

Python 3 and Python 2 have significant incompatibilities that would make it quite challenging to manually maintain versions of controller.py for each of them (and it would also be somewhat challenging to manually keep track of the slight differences between different recent versions of 3).  If we're doing this manually, I think it would be best to pick one version and stick with it, perhaps occasionally incrementing the Python version requirement if we want to make use of new features.

Type-hinting in Python didn't become a thing until around version 3.5 (circa 2015), and it has received notable improvements in more recent releases, including in version 3.10 that was just released this month.  

The "pythonic improvements" that I'd been hoping to include likely also rely on having a somewhat recent version of Python 3, and I'm not sure the extent to which they'd even be possible in Python 2.  Even if they are possible, the conversion would probably require more work than I'd want to put in.

So, anyway, the fewer versions of Python you support, the greater the risk of losing some people who are unwilling to install and work with a newer version.  But the more versions you want to cater to, the greater the challenge would be of manually updating versions of controller.py to go with each of them.  In particular, my own willingness to contribute probably wouldn't extend much further than making it work with some specific recent version of Python3, e.g. 3.8 or even 3.10.

PeterJCLaw, 2021-10-16:
Looking at the public docs, it seems only Python 3.7+ is supported? So Python 2 and 3.5/3.6 aren't really concerns any more.

While there definitely are new features in more recent versions (deferred annotation processing being the main one), I'm not aware of any big changes that would be necessary to support specifically that would cause issues for supporting 3.7+.

Lazy annotations can even be enabled via `__future__` import, but IMO are nice to have rather than being particularly arduous to work without.

omichel, 2021-10-16:
Sure, we can fully drop Python 2 support.

omichel, 2021-10-16:
I believe this new python API should be implemented from scratch. Basically, it should simply load the libController shared library and call it's functions wherever needed.

Justin-Fisher, 2021-10-16:
Sorry if I'm being slow/dense on this.  Are you talking about libController.a?  Do you know how to import that in Python?  And how to access the relevant functions in Python?  Are the available functions exactly the ones that Webots docs list for the C language?  (I may be wrong, but it seems like the current controller.py calls some API functions that I didn't think were documented.  Maybe they were constructed by swig?)

My very strong inclination would be to start with something that works, like controller.py and at least use it as a checklist of things that should probably somehow have analogs in the new module (aside from some pretty-clearly-useless bloat that swig added).

I actually did embark on this today, and have done a fair amount of work streamlining the controller.py device creation process, in a way that should make it pretty easy to switch API's, I think, though I still haven't wrapped my head around what exactly will be different about the C-API.  For now, I'm still having it use the swig/C++ API, since I know well enough how to use it and can test that my changes haven't completely broken things.  So far so good...

omichel, 2021-10-17:
I believe developing a python module that loads a C shared library and calls its functions is something pretty standard. Yes, you need to load libController which is named with a .dll extension on Windows, .so on Linux and .dylib on macOS, but I guess there is a generic way to achieve this. Then, yes, you should call the C API functions which are documented in the reference manual. If you cannot manage to do it, I could create a working skeleton next week into which it should be easy to port your code.

omichel, 2021-10-17:
Sorry, I meant `controller.dll` indeed. I will give the C implementation a go next week and keep you posted.

Justin-Fisher, 2021-10-17:
I don't suppose you know already what type the "WbDeviceTag" returned by `wb_robot_get_device` will have when it gets passed from the C API to python?  Just an integer?

omichel, 2021-10-17:
I believe it is an opaque C pointer, but can be casted to an integer anyhow.

Justin-Fisher, 2021-10-17:
Similarly, the old `controller.py` includes many duplicate functions that are defined both as documented static methods of classes, e.g. `GPS.convertToDegreesMinutesSeconds` and as undocumented stand-alone functions like `GPS_convertToDegreesMinutesSeconds`.  This amounts to having two different names for the very same thing.  In general, the documented version would probably be preferred programming practice.  Is it fine to abandon the undocumented versions, or should they be kept for backwards compatibility?

omichel, 2021-10-18:
Yes, that should be easy to handle by introducing specific cases for Linux/macOS/Windows. For now, you can go ahead and consider that we will take care of this later.

omichel, 2021-10-20:
I wouldn't use class names as a replacement for constants. As you mentioned, it might be quite confusing and wouldn't work for all constants. But I like you idea of defining these constants in the DLL rather than in include files. I will think on what the best way to implement it.

Justin-Fisher, 2021-10-20:
How much assurance is Webots offering that constants will retain their values from Webots version to Webots version?  If it is assured that constants will retain their values, then (1) that makes it easier to create a one-stop repository for constant values, and (2) there may be some advantage to leaving them as integers since integers could be used in some mathematical ways, e.g. as indices of arrays, or in boolean evaluations.  

If, on the other hand, there's no assurance that constants will retain their values from webots version to version (e.g., if a new item might get inserted in the middle of an enumerated list, and if the expectation is that these constants will be useful only for identity comparisons) then (1) we'd probably want to automate, as much as possible, tracing back of constant names to their current values to avoid the risk of failing to keep up with a change, and (2) we'd be more free to assign new values, like informative strings, to these constants in some languages.

ygoumaz, 2022-11-08:
@omichel Is this issue still relevant now that the Python API has been implemented?

omichel, 2022-11-08:
Type annotations were introduced in the new Python interface. However, they are not yet used in the Python API documentation, which would be nice. So, let's keep this issue open for now.

ygoumaz, 2022-11-08:
Closing as duplicate of #2959.

closed 2022-11-08 by ygoumaz

PeterJCLaw, 2022-11-08:
Aha, nice. Is there an example of using the types to check client code? Are the types published on PyPI or just as part of the main webots install? (Apologies if I've missed some docs on this)

omichel, 2022-11-08:
Those are just type annotations in the source code of the python module. It is part of the main Webots install and not published on PyPI.

PeterJCLaw, 2022-11-08:
Would you be open to publishing stubs on PyPI? I'm not actually sure how to do it, but I'm imagining a tool extracting the stubs from the source and then building a package around them. That package could then be used in CI etc. where a full install would be quite heavyweight.
(If that's interesting I'd be happy to open a separate issue for it)

omichel, 2022-11-09:
Yes, that is a good idea. However, since the Python API depends on the shared library binaries of the libController, it might be tricky to set it up properly on PyPI and support all architectures (Windows, Linux and macOS). But if you are willing to try this out, feel free to explore it and let us know your findings. Eventually, we would be happy to review your PR introducing such a contribution.

PeterJCLaw, 2022-11-11:
Have done some experimenting; the results are in https://github.com/cyberbotics/webots/pull/5472 though mostly as a PR seemed a useful way to share the results.