cyberbotics/webots #2386: Movie recording is cancelled if simulation is reset before movie starts

PeterJCLaw, 2020-10-25:
**Describe the Bug**

If I reset the simulation before starting the movie capture then the capture is always cancelled.

**Steps to Reproduce**

I'm running `webots` with `webots --batch worlds/Arena.wbt`, running without `--batch` results in the message being a dialog box rather than a log message, but has the same effect.

Controller:
``` python
from controller import Supervisor
supervisor = Supervisor()

# I also pause the simulation here, though that doesn't appear to affect the bug

supervisor.simulationReset()

# I also restart all the other (non-supervisor) controllers at this point too,
# though that doesn't appear to affect the bug

supervisor.movieStartRecording(
    'that.mp4', width=640, height=480, quality=100,
    codec=0, acceleration=1, caption=False,
)
# Short step length for easy reproduce
supervisor.step(320)
supervisor.movieStopRecording()
supervisor.simulationSetMode(Supervisor.SIMULATION_MODE_PAUSE)
```

This results in a log line of `INFO: Video creation canceled.`.
(aside: there's a typo in "canceled")

Commenting out the `simulationReset` call successfully captures the video.

**Expected behavior**

I would like this to successfully record the video.

**System**
 - Operating System: Ubuntu 18.04.5 LTS

**Additional context**

My goal here is to be able to capture a video of the whole simulation, from the very start of all the (non-supervisor) controllers, hence the desire to pause then reset the simulation before starting the movie.
(I've also recently found the `--mode=pause` argument, which may provide an alternative for my case, though I think the above is still a bug)

PeterJCLaw, 2020-10-25:
> I've also recently found the `--mode=pause` argument, which may provide an alternative for my case

Turns out that this doesn't help me because of course when starting paused the supervisor controller doesn't get started upfront.

omichel, 2020-10-26:
That is clearly a bug.
However, if you add a call to `supervisor.step(32)` just after the `supervisor.simulationReset()`, it fixes the problem.

omichel, 2020-10-26:
BTW, "canceled" is correct in American English, which is what we use in Webots.

stefaniapedrazzi, 2020-10-28:
The `Supervisor.simulationReset()` method is applied only at the end of the step (see [documentation](https://www.cyberbotics.com/doc/reference/supervisor?tab-language=python#wb_supervisor_simulation_reset)) to guarantee that the simulation works correctly even if multiple Robot and Supervisor controllers are running.
On the other hand the `Supervisor.movieStartRecording` method is applied immediately, i.e. with the posted code the movie recording is started before the reset.

So as @omichel suggested, calling `supervisor.step(32)` just after the `supervisor.simulationReset()` would make sure that reset is called before initializing the movie and that the first frame of the movie will record the world state after reset (and not before the reset).

In #2406 I'm fixing the simulation reset behavior so that the movie recording is not canceled on simulation reset but only when the Supervisor controller that requested the recording is restarted.

PeterJCLaw, 2020-10-28:
> The `Supervisor.simulationReset()` method is applied only at the end of the step (see [...]

Ah, thanks. I had missed that detail, though I suspect I would also have missed the subtleties of the interaction between reset and `movieStartRecording` even if I had seen it. Thanks for laying this out clearly.
Aside: starting & stopping an animation at the same place as I am a movie does work already -- are the semantics different for the animation than the movie?

> So as @omichel suggested, calling `supervisor.step(32)` just after the `supervisor.simulationReset()` would make sure that reset is called before initializing the movie and that the first frame of the movie will record the world state after reset (and not before the reset).

To check my understanding of this -- would doing this mean that the other controllers would have a chance to operate on the world after the reset but before the first frame of the movie? (In my original I'm actually restarting them after the reset but before the movie starts, though I'm not sure that will affect the semantics)

I've actually now got an alternative solution which uses `customData` and cooperative controllers to avoid needing to reset the world at all, however the semantics could still be useful to understand.

> In #2406 I'm fixing the simulation reset behavior so that the movie recording is not canceled on simulation reset but only when the Supervisor controller that requested the recording is restarted.

:+1:

stefaniapedrazzi, 2020-10-29:
> Aside: starting & stopping an animation at the same place as I am a movie does work already -- are the semantics different for the animation than the movie?

The difference is simply that on simulation reset the animation is not canceled even if started from a Supervisor controller.

> To check my understanding of this -- would doing this mean that the other controllers would have a chance to operate on the world after the reset but before the first frame of the movie? (In my [...]

Other controllers will execute all the operations listed before the first `robot.step()` call after the simulation reset, in your case all the operations before the first `robot.step()` instruction. But on simulation reset these operations should be reverted and if you [restart the other controllers](https://www.cyberbotics.com/doc/reference/supervisor?tab-language=python#wb_supervisor_node_restart_controller) just after resetting the simulation you should be able to record every operation.

stefaniapedrazzi, 2020-10-29:
It should be fixed in #2406 and available tomorrow in the latest nightly build
@PeterJCLaw please reopen the issue if it won't work for you.