cyberbotics/webots #2387: Internal AttributeError when accessing invalid attribute in Python

PeterJCLaw, 2020-10-25:
**Describe the Bug**

If you happen to typo the name of a method or attribute in your Python controller you get an error which appears to come from the Webots controller rather than your code.

**Steps to Reproduce**

``` python
from controller import Robot

Robot().oops
```

results in:

``` traceback
Traceback (most recent call last):
  File "demo.py", line 3, in <module>
    Robot().oops
  File "/usr/local/webots/lib/controller/python36/controller.py", line 1701, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, Node, name)
  File "/usr/local/webots/lib/controller/python36/controller.py", line 88, in _swig_getattr
    return _swig_getattr_nondynamic(self, class_type, name, 0)
  File "/usr/local/webots/lib/controller/python36/controller.py", line 83, in _swig_getattr_nondynamic
    return object.__getattr__(self, name)
AttributeError: type object 'object' has no attribute '__getattr__'
```

**Expected behavior**

While `AttributeError` is correct here, I would expect that it would have the proper name of the attribute which was being attempted to access (i.e: `oops` in the above).

**System**
 - Operating System: Ubuntu 18.04

**Additional context**

I'm not sure what the code in Webots' `controller.py` is trying to do, but as far as I can determine `object.__getattr__` is always going to be an error -- there isn't a class method of that name in any version of Python.

If this is meant to always error, then a simple fix would be to change `_swig_getattr_nondynamic` like so:
``` diff
-    if (not static):
-        return object.__getattr__(self, name)
-    else:
-        raise AttributeError(name)
+    raise AttributeError(name)
```
Alternatively, perhaps the function `getattr` is intended to be being used?
``` diff
     if (not static):
-        return object.__getattr__(self, name)
+        return getattr(self, name)
     else:
         raise AttributeError(name)
```

omichel, 2020-10-26:
That's probably an issue caused by SWIG, yet another good reason to create a Python API which doesn't rely on SWIG, see also #2385 about it.

ygoumaz, 2022-11-08:
Closing this issue as it was fixed with the new Python API.

closed 2022-11-08 by ygoumaz