cyberbotics/webots #2417: Angular acelleration behaviour in differential drive robot

Prvg95, 2020-10-29:
I am using Matlab to as a extern controller for the robot and when i give opposite torques to the motor in order to make him rotate with constant angular acceleration the accelearation tends to have an oscillatory behaviour and decrases for bigger values of angular velocity. This effect tends to decrease the higher i put the CFM value in contact properties. Thank you very much

**Steps to Reproduce**
1. Run the simulation with the torques set for 0.07 and -0.07 for each side motors 
2. Observe the sensor information from the gyroscope 
3. Integrate the angular velocity given ( I am using Matlab for this) and check the behaviour.

**Expected behavior**
I expected the angular acceleration to be kept constant with constant torques and not decraese or oscilate.

**Screenshots**
This is the image with the default CFM
[image]

[Cher2.3.zip](https://github.com/cyberbotics/webots/files/5458929/Cher2.3.zip)

**System**
 - Operating System:  Windows 10
 - Graphics Card:  Nvidia GeForce GTX 1050 Ti

DavidMansolino, 2020-10-30:
Thank you for this precise bug report.
Just to be sure, you are using the latest version of Webots (R2020b-rev1) right?

Prvg95, 2020-10-30:
Hello i am currently still using the version R2020a revision 1. Does the bug not occur in the new version? If so i am sorry for the miscommunication. I will update my version and see if the behavior continues

Prvg95, 2020-10-30:
I have checked in the new version of Webots and the same behaviour happens

lukicdarkoo, 2021-03-02:
> Integrate the angular velocity given ( I am using Matlab for this) and check the behaviour.

If you integrate the angular velocity you get an angle, not acceleration. Did you calculate integral or derivative?

To help us, could you please create an example in Python, so we can easily test it?

akshattodi, 2021-03-26:
Can anyone tell me how to get the values of gyroscope x, y, z, axis values with python in webots . I have some problems in values of gyroscope . I get the same values when the robot is stand in perfect straight condition at one place and when the root is moving with some tilt , the values that I get are ([-0.07224864573101453, 1.7716707096747686, -0.041035580080441614], [1.3429357188987567, -0.7797925328713822, 0.8719042901146818], [2.3055278837609823, 1.849134301769992, 1.183000112299603], [1.463396781593345, 2.3795468577022043, 1.9369502530507368], [0.37185974993731186, 1.9258312651014604, 1.844386392125862], .... etc) these values are in rad/sec, and after converting it into degrees ([-4, 101, -2], [76, -44, 49], [132, 106, 67], [83, 136, 111], [21, 110, 105], .... etc) . The values are always fix.

### **My Code**
    
    """drive_my_robot controller."""
    
    # You may need to import some classes of the controller module. Ex:
    #  from controller import Robot, Motor, DistanceSensor
    
    from controller import Robot
    
    
    if __name__ == "__main__":
        
        # create teh Robot instance.
        robot = Robot()
        
        # get the time step of the current world.
        timestep = 100
        max_speed = 16.28
        
        gyrosc = robot.getDevice('gyro')
        gyrosc.enable(timestep)
        
        # Created motor instances
        left_motor = robot.getDevice('motor_1')
        right_motor = robot.getDevice('motor_2')
        
        left_motor.setPosition(float('inf'))
        left_motor.setVelocity(0.0)
        
        right_motor.setPosition(float('inf'))
        right_motor.setVelocity(0.0)
        
        count = 0
        
        # Main loop:
        # - perform simulation steps until Webots is stopping the controller
        while robot.step(timestep) != -1:
        
            print('Hello world')
            
            left_speed = 0.5 * max_speed
            right_speed = 0.5 * max_speed
            
            
            # Converting radian values into degrees
            gyrovalues = gyrosc.getValues()
            gyroX = int((gyrovalues[0] * 180)/(3.14))
            gyroY = int((gyrovalues[1] * 180)/(3.14))
            gyroZ = int((gyrovalues[2] * 180)/(3.14))
            gyrodeg = [gyroX, gyroY, gyroZ]
            
            count += 1
            print(count)
            #print(gyrosc.getSamplingPeriod())
            #print(gyrovalues[0])
            #print(int(gyroX))
            print(gyrodeg)
            print(gyrosc.getValues())
            print(gyrosc.getLookupTable())
               
            left_motor.setVelocity(left_speed)
            right_motor.setVelocity(right_speed)
                
            pass
        
        # Enter here exit cleanup c*ode.

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel